gbionics/jaxsim #27: Solve unresolved dependency on jaxlib

fl-ferr, 2023-04-14:
In releases 0.1.dev117 and [0.1.dev118](https://pypi.org/project/jaxsim/) there is a strict dependency on [jaxlib](https://pypi.org/project/jaxlib/)==0.3.15 which is not currently available, making it impossible to install via PyPI

diegoferigo, 2023-04-14:
This is one of the longstanding issues of jaxsim. I did most of the development in late 2021 / early 2022 when `jax.experimental.loops` was available. Recent `jax` versions removed it. As long as I don't find enough time to switch to `jax.lax.scan`, we cannot update `jax`.

Here you can find the occurrences that have to be migrated (it's no so straightforward unfortunately): https://github.com/search?q=repo%3Aami-iit%2Fjaxsim+jax.experimental.loops&type=code.

Until the migration, we need to use jax and jaxlib 0.3.15. Why do you say that it is not available in PyPI? What Python version are you using?

See all the PyPI wheels at https://pypi.org/project/jaxlib/0.3.15/#files.

fl-ferr, 2023-04-14:
Thanks for your answer, I will have a look at it. I'm using Python 3.10

diegoferigo, 2023-04-18:
For those interested in the current status of the repository, the Docker image at the following link provides a working setting where JaxSim can run.

https://gist.github.com/diegoferigo/d8c75132a3193b402514d33f6e6240b1

At the time of writing, the corresponding conda environment is the one below.

<details>
<summary>environment.yml</summary>

```yaml
name: base
channels:
  - conda-forge
  - defaults
dependencies:
  - _libgcc_mutex=0.1
  - _openmp_mutex=4.5
  - abseil-cpp=20211102.0
  - absl-py=1.4.0
  - aiohttp=3.7.4.post0
  - alsa-lib=1.2.8
  - anyio=3.6.2
  - argon2-cffi=21.3.0
  - argon2-cffi-bindings=21.2.0
  - asttokens=2.2.1
  - astunparse=1.6.3
  - async-timeout=3.0.1
  - attr=2.5.1
  - attrs=22.2.0
  - backcall=0.2.0
  - backports=1.0
  - backports.functools_lru_cache=1.6.4
  - beautifulsoup4=4.12.2
  - binutils=2.39
  - binutils_impl_linux-64=2.39
  - binutils_linux-64=2.39
  - bleach=6.0.0
  - blinker=1.6.2
  - boltons=23.0.0
  - brotli=1.0.9
  - brotli-bin=1.0.9
  - brotlipy=0.7.0
  - bzip2=1.0.8
  - c-ares=1.18.1
  - c-compiler=1.5.2
  - ca-certificates=2022.12.7
  - cached-property=1.5.2
  - cached_property=1.5.2
  - cachetools=5.3.0
  - cairo=1.16.0
  - certifi=2022.12.7
  - cffi=1.15.1
  - chardet=4.0.0
  - charset-normalizer=3.1.0
  - click=8.1.3
  - cloudpickle=2.2.1
  - colorama=0.4.6
  - coloredlogs=15.0.1
  - comm=0.1.3
  - commonmark=0.9.1
  - compilers=1.5.2
  - conda=23.3.1
  - conda-package-handling=2.0.2
  - conda-package-streaming=0.7.0
  - contourpy=1.0.7
  - cryptography=39.0.0
  - cuda-cccl_linux-64=12.0.90
  - cuda-cudart=12.0.107
  - cuda-cudart-dev=12.0.107
  - cuda-cudart-dev_linux-64=12.0.107
[... 310 lines cut ...]
  - xorg-xproto=7.0.31
  - xz=5.2.6
  - yaml=0.2.5
  - yaml-cpp=0.7.0
  - yarl=1.8.2
  - zeromq=4.3.4
  - zipp=3.15.0
  - zlib=1.2.13
  - zstandard=0.19.0
  - zstd=1.5.2
  - pip:
      - chex==0.1.6
      - distrax==0.1.3
      - jax-dataclasses==1.3.0
      - jaxlie==1.2.10
      - meshcat==0.3.2
      - meshcat-viz==0.1.dev38
      - overrides==7.3.1
      - pyngrok==6.0.0
      - rod==0.1.dev79
      - u-msgpack-python==2.7.2
      - xmltodict==0.13.0
```

</details>

fl-ferr, 2023-06-15:
This will be solved by #35

closed 2023-06-30 by diegoferigo