gbionics/jaxsim #41: Implement algorithm to compute the standalone Coriolis matrix `C(q, ν)`

diegoferigo, 2023-06-30:


traversaro, 2023-06-30:
A possible algorithm to implement is https://arxiv.org/pdf/2010.01033 .

diegoferigo, 2023-06-30:
Thanks! I already had a look at it. Another interesting xref is https://github.com/robotology/idyntree/issues/745.

flferretti, 2023-11-30:
Check out: [flferretti/jaxsim @ feature/coriolis](https://github.com/flferretti/jaxsim/tree/feature/coriolis)

diegoferigo, 2024-06-10:
Now that #169 added the support of computing $\dot{J}$, I tried to compute the Coriolis matrix without implementing a new RBDA (I still have few doubts on how to handle the base-related blocks for an iterative algorithm). The computation is pretty slow, but downstream users that need this matrix for particular controllers at least can play with it.

I'll leave this issue open since an iterative algorithm is desired. Maybe someone, now that there is a ground truth, is willing enough to dig into this implementation sooner or later.