gbionics/jaxsim #60: Write notebook to demonstrate use of GPU simulation of model with PD+GravityCompensation or inverse kinematics

traversaro, 2023-11-29:
First step:
* Show simulation
* Possibly show how train a policy

As a first round, we can probably use a fixed based robot.

flferretti, 2023-11-29:
Check out the `Development` section for the related working branch

traversaro, 2023-12-12:
I tried to add a visualization based on iDynTree, but I forgot that in iDynTree on PyPI I disabled the irrlicht deps in https://github.com/robotology/idyntree/pull/1068, so I get as error:
~~~
[ERROR] ModelLoader :: loadModelFromFile : Error in parsing model from URDF.
[ERROR] Visualizer :: init : Impossible to use iDynTree::Visualizer, as iDynTree has been compiled without Irrlicht.
[ERROR] Visualizer :: addModel : Impossible to use iDynTree::Visualizer, as iDynTree has been compiled without Irrlicht.
[ERROR] Visualizer :: drawToFile : Impossible to call drawToFile in a not initialized visualizer
~~~

flferretti, 2023-12-12:
We could try to use the conda version of IDynTree using [`condacolab`](https://github.com/conda-incubator/condacolab). Otherwise, tomorrow I will work on the use of `pixi` to make the example easier to use

flferretti, 2023-12-13:
In https://github.com/ami-iit/jaxsim/commit/856d06e4a6a58e6e08fdf12d7fde83c50b804485 I added `pixi` support for `linux-64` and `osx-64`, where the latter uses [JAX Metal](https://developer.apple.com/metal/jax/) for GPU support on Apple Silicon. As it is still in experimental phase, we can think of dropping support for the examples on OSX, for the time being

flferretti, 2024-01-09:
Notebook added in #63. Closing

closed 2024-01-09 by flferretti