gbionics/jaxsim #76: Add comprehensive testing strategy for real-world deployment on iCub and ErgoCub

flferretti, 2024-02-02:
The current testing framework for `jaxsim` lacks validation against the actual robots intended for deployment. This issue aims to address this gap by establishing a comprehensive testing strategy that ensures the reliability and performance of the simulator in real-world scenarios.

> Perhaps we can test against also the iCub/ergoCub models, as most probably those are the one we are going to use the feature against?
> 
> Kind of unrelated to the PR, but just because I realized it: testing real world models is ok, but in my experience you may have some blinds spots. In iDynTree I historically found many problems with programmatically generated random models, generated by [`getRandomModel`](https://github.com/robotology/idyntree/blob/4a981e11e282ab5c10009523157ba151b2519509/src/model/include/iDynTree/ModelTestUtils.h#L123) function. 
> 
> It could make sense to do something similar here, or also directly use it here as we can export the randomly generated model to URDF, something like
> ~~~python
> import idyntree.bindings as idt
> nrOfJoints = 20
> randomModel = idt.getRandomModel(nrOfJoints)
> mdlExport = idt.ModelExporter()
> mdlExport.init(randomModel)
> urdfString = "dummy"
> urdfFilePath = "/dummy/dummy"
> mdlExport.exportModelToString(urdfString)
> # or
> mdlExport.exportModelToFile(urdfFilePath)
> ~~~

_Originally posted by @traversaro in https://github.com/ami-iit/jaxsim/pull/75#discussion_r1475776719_

diegoferigo, 2024-02-02:
I agree about testing on iCub/ErgoCub models, but note that the JIT performance we currently have would make the tests take really a lot of time. This was the original reason to test smaller fixed-base and floating-base models.

traversaro, 2024-02-02:
> I agree about testing on iCub/ErgoCub models, but note that the JIT performance we currently have would make the tests take really a lot of time, and they are already very slow. This was the original reason to test smaller fixed-base and floating-base models.

Ah, good point. I was not considering that.

diegoferigo, 2024-03-11:
In #106, I'm testing -among other models- a reduced version of the ErgoCub robot. It could be a good tradeoff between CI time and comprehensiveness of the testing strategy.

diegoferigo, 2024-03-11:
Furthermore, random models could be created with `rod` similarly to the existing `box` and `sphere` models.

diegoferigo, 2024-03-27:
The testing suite targeting the new functional APIs test against a reduced version of ErgoCub (only the DoFs of the legs are kept). It seems a acceptable trade-off between the comprehensiveness of the tests and the time necessary to run them (~20 mins at the time of writing).

I guess we can close this issue now.