gbionics/jaxsim #103: Error when passing two different pytrees with the same structure to a JIT-compiled function

diegoferigo, 2024-03-07:
Also with the new functional APIs, the following problem still affects our code:

```
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
```

A MWE to reproduce the problem is the following:

```python
import jax.numpy as jnp
import jaxsim.api as js
import rod.builder.primitives
import rod.urdf.exporter

# Create on-the-fly a ROD model of a box.
rod_model = (
    rod.builder.primitives.BoxBuilder(x=0.3, y=0.2, z=0.1, mass=1.0, name="box")
    .build_model()
    .add_link()
    .add_inertial()
    .add_visual()
    .add_collision()
    .build()
)

# Export the URDF string.
urdf_string = rod.urdf.exporter.UrdfExporter.sdf_to_urdf_string(
    sdf=rod_model, pretty=True
)

model1 = js.model.JaxSimModel.build_from_model_description(
    model_description=urdf_string,
    gravity=jnp.array([0, 0, -10]),
    is_urdf=True,
)

model2 = js.model.JaxSimModel.build_from_model_description(
    model_description=urdf_string,
    gravity=jnp.array([0, 0, -10]),
    is_urdf=True,
)

# This works
_ = js.contact.estimate_good_soft_contacts_parameters(model=model1)

# This raises
_ = js.contact.estimate_good_soft_contacts_parameters(model=model2)
```

I've already tried to play a bit with `JaxSimModel.__eq__` and `JaxSimModel.__hash__` with no luck.
Marking this issue as help wanted in case this problem becomes annoying to someone.

Interestingly, running it in a context in which `jax.jit` is disabled, works fine:

```python
with jax.disable_jit():
    _ = js.contact.estimate_good_soft_contacts_parameters(model=model1)
    _ = js.contact.estimate_good_soft_contacts_parameters(model=model2)
```

I suspect that there is something strange in the jax cache that we are either overlooking or not handling properly.

diegoferigo, 2024-03-07:
This is basically the updated version of the problem occurring in the previous OOP APIs in #84 cc @flferretti

traversaro, 2024-03-08:
The example is failing with the following error for me:
~~~
traversaro@IITBMP014LW012:~/jaxsimws$ pixi run python mwe103.py
jaxsim[21436] INFO Enabling JAX to use 64bit precision
An NVIDIA GPU may be present on this machine, but a CUDA-enabled jaxlib is not installed. Falling back to cpu.
Traceback (most recent call last):
  File "/home/traversaro/jaxsimws/mwe103.py", line 18, in <module>
    urdf_string = rod.urdf.exporter.UrdfExporter.sdf_to_urdf_string(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/traversaro/jaxsimws/.pixi/envs/default/lib/python3.12/site-packages/rod/urdf/exporter.py", line 30, in sdf_to_urdf_string
    model = sdf.models()[0]
            ~~~~~~~~~~~~^^^
IndexError: list index out of range
~~~

Installed packages:
<details>

<summary>Click</summary>

~~~
traversaro@IITBMP014LW012:~/jaxsimws$ pixi list
Package                               Version       Build                  Size       Kind   Source
_libgcc_mutex                         0.1           conda_forge            2.5 KiB    conda  _libgcc_mutex-0.1-conda_forge.tar.bz2
_openmp_mutex                         4.5           2_gnu                  23.1 KiB   conda  _openmp_mutex-4.5-2_gnu.tar.bz2
_sysroot_linux-64_curr_repodata_hack  3             h69a702a_14            20.7 KiB   conda  _sysroot_linux-64_curr_repodata_hack-3-h69a702a_14.conda
absl-py                               2.1.0         pyhd8ed1ab_0           104.8 KiB  conda  absl-py-2.1.0-pyhd8ed1ab_0.conda
alsa-lib                              1.2.11        hd590300_1             541.7 KiB  conda  alsa-lib-1.2.11-hd590300_1.conda
ampl-mp                               3.1.0         h2cc385e_1006          1.1 MiB    conda  ampl-mp-3.1.0-h2cc385e_1006.tar.bz2
aom                                   3.8.1         h59595ed_0             2.6 MiB    conda  aom-3.8.1-h59595ed_0.conda
assimp                                5.3.1         hfb0e8fe_2             3.4 MiB    conda  assimp-5.3.1-hfb0e8fe_2.conda
asttokens                             2.4.1         pyhd8ed1ab_0           28.2 KiB   conda  asttokens-2.4.1-pyhd8ed1ab_0.conda
atk-1.0                               2.38.0        hd4edc92_1             539 KiB    conda  atk-1.0-2.38.0-hd4edc92_1.tar.bz2
attr                                  2.5.1         h166bdaf_1             69.4 KiB   conda  attr-2.5.1-h166bdaf_1.tar.bz2
aws-c-auth                            0.7.16        h79b3bcb_6             101 KiB    conda  aws-c-auth-0.7.16-h79b3bcb_6.conda
aws-c-cal                             0.6.10        hb29e0c7_1             53.8 KiB   conda  aws-c-cal-0.6.10-hb29e0c7_1.conda
aws-c-common                          0.9.13        hd590300_0             220.3 KiB  conda  aws-c-common-0.9.13-hd590300_0.conda
aws-c-compression                     0.2.18        hecc5fa9_1             18.7 KiB   conda  aws-c-compression-0.2.18-hecc5fa9_1.conda
aws-c-event-stream                    0.4.2         hf9b2f7b_4             52.7 KiB   conda  aws-c-event-stream-0.4.2-hf9b2f7b_4.conda
aws-c-http                            0.8.1         h5d7533a_5             190.5 KiB  conda  aws-c-http-0.8.1-h5d7533a_5.conda
aws-c-io                              0.14.5        h50678d4_1             153.5 KiB  conda  aws-c-io-0.14.5-h50678d4_1.conda
aws-c-mqtt                            0.10.2        hf479d2b_4             160.6 KiB  conda  aws-c-mqtt-0.10.2-hf479d2b_4.conda
aws-c-s3                              0.5.2         h4ad9680_0             102.8 KiB  conda  aws-c-s3-0.5.2-h4ad9680_0.conda
aws-c-sdkutils                        0.1.15        hecc5fa9_1             54.1 KiB   conda  aws-c-sdkutils-0.1.15-hecc5fa9_1.conda
aws-checksums                         0.1.18        hecc5fa9_1             49 KiB     conda  aws-checksums-0.1.18-hecc5fa9_1.conda
aws-crt-cpp                           0.26.2        h19f5d62_7             326 KiB    conda  aws-crt-cpp-0.26.2-h19f5d62_7.conda
aws-sdk-cpp                           1.11.267      h5606698_1             3.4 MiB    conda  aws-sdk-cpp-1.11.267-h5606698_1.conda
azure-core-cpp                        1.11.1        h91d86a7_1             334.6 KiB  conda  azure-core-cpp-1.11.1-h91d86a7_1.conda
azure-storage-blobs-cpp               12.10.0       h00ab1b0_1             505 KiB    conda  azure-storage-blobs-cpp-12.10.0-h00ab1b0_1.conda
azure-storage-common-cpp              12.5.0        h94269e2_4             129.3 KiB  conda  azure-storage-common-cpp-12.5.0-h94269e2_4.conda
binutils_impl_linux-64                2.40          hf600244_0             5.2 MiB    conda  binutils_impl_linux-64-2.40-hf600244_0.conda
binutils_linux-64                     2.40          hbdbef99_2             27.5 KiB   conda  binutils_linux-64-2.40-hbdbef99_2.conda
blosc                                 1.21.5        h0f2a231_0             47.6 KiB   conda  blosc-1.21.5-h0f2a231_0.conda
brotli                                1.1.0         hd590300_1             18.9 KiB   conda  brotli-1.1.0-hd590300_1.conda
brotli-bin                            1.1.0         hd590300_1             18.5 KiB   conda  brotli-bin-1.1.0-hd590300_1.conda
bullet-cpp                            3.25          hfb8ada1_2             40.4 MiB   conda  bullet-cpp-3.25-hfb8ada1_2.conda
bzip2                                 1.0.8         hd590300_5             248.3 KiB  conda  bzip2-1.0.8-hd590300_5.conda
c-ares                                1.27.0        hd590300_0             159.7 KiB  conda  c-ares-1.27.0-hd590300_0.conda
[... cut ...]
zipp                                  3.17.0        pyhd8ed1ab_0           18.5 KiB   conda  zipp-3.17.0-pyhd8ed1ab_0.conda
zlib                                  1.2.13        hd590300_5             90.6 KiB   conda  zlib-1.2.13-hd590300_5.conda
zstd                                  1.5.5         hfc55251_0             532.4 KiB  conda  zstd-1.5.5-hfc55251_0.conda
zziplib                               0.13.69       h27826a3_1             96.8 KiB   conda  zziplib-0.13.69-h27826a3_1.tar.bz2
~~~

</details>

Perhaps a specific rod version is required?

diegoferigo, 2024-03-08:
I guess it needs https://github.com/ami-iit/rod/pull/27/commits/0ce60b22e18432f6a557e83f2a1100cb0bd4cdd4 included in https://github.com/ami-iit/rod/pull/27.

diegoferigo, 2024-03-08:
Also #83 actually needed https://github.com/ami-iit/rod/pull/27 for proper color support, we need to bump the dependencies before releasing jaxsim `v0.2.0`.

traversaro, 2024-03-09:
Playing a bit around, I was able to reduce the MWE, to remove jaxsim at all:

~~~python
import dataclasses

import jax
import jax.numpy as jnp
import jax_dataclasses
from jax_dataclasses import Static
import pathlib
import numpy.typing as npt

import dataclasses
from typing import Dict, Union

import jax.lax
import jax.numpy as jnp
import jax_dataclasses
import numpy as np
from jax_dataclasses import Static

@jax_dataclasses.pytree_dataclass
class MWE103GroundContact:
    body: Static[npt.NDArray] = dataclasses.field(
        default_factory=lambda: np.array([])
    )

    @staticmethod
    def build_from() -> "MWE103GroundContact":
        # Build the object
        gc = MWE103GroundContact(body=np.zeros(8))
        return gc

def local_total_mass(model: MWE103GroundContact) -> float:
    return 1.0

dummy_contact1 = MWE103GroundContact.build_from()
dummy_contact2 = MWE103GroundContact.build_from()

local_total_mass_jit1 = jax.jit(local_total_mass)
local_total_mass_jit2 = jax.jit(local_total_mass)

_ = local_total_mass_jit1(model=dummy_contact1)
_ = local_total_mass_jit2(model=dummy_contact2)
~~~

Somehow the problem is related to the `body` attribute of the `GroundContact` class: https://github.com/ami-iit/jaxsim/blob/f3d9abb0dd8583c5391db7b47c998ad0a457adea/src/jaxsim/physics/model/ground_contact.py#L26 . Commenting it, the problem disappears.

traversaro, 2024-03-09:
> Somehow the problem is related to the `body` attribute of the `GroundContact` class:

This is aligned with what was found in https://github.com/ami-iit/jaxsim/issues/84#issuecomment-1943719169 .

traversaro, 2024-03-09:
I think the answer to this question clarifies everything: https://github.com/google/jax/issues/19547#issuecomment-1913711588 .

traversaro, 2024-03-10:
I think I described the root issue behind all the `ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()` in https://github.com/ami-iit/jaxsim/pull/105 .

flferretti, 2024-04-16:
Doing some additional tests, I found out that a similar problem is raised. This time, it seems to be related to the integrators:

```py
Traceback (most recent call last):
  File "/home/flferretti/git/comodo/src/comodo/jaxsimSimulator/test.py", line 55, in <module>
    integrator_state2 = integrator2.init(x0=data2.state, t0=0, dt=1e-3)
  File "/home/flferretti/jaxsim/src/jaxsim/integrators/common.py", line 166, in init
    _ = integrator(x0, t0, dt, **kwargs)
  File "/home/flferretti/jaxsim/src/jaxsim/integrators/common.py", line 283, in __call__
    z = self._compute_next_state(x0=x0, t0=t0, dt=dt, **kwargs)
  File "/home/flferretti/jaxsim/src/jaxsim/integrators/common.py", line 417, in _compute_next_state
    K, _ = jax.lax.scan(
  File "/home/flferretti/jaxsim/src/jaxsim/integrators/common.py", line 403, in scan_body
    ki = jax.lax.cond(
  File "/home/flferretti/jaxsim/src/jaxsim/integrators/common.py", line 374, in <lambda>
    get_ẋ0 = lambda: self.params.get("dxdt0", f(x0, t0)[0])
  File "/home/flferretti/jaxsim/src/jaxsim/integrators/common.py", line 365, in <lambda>
    f = lambda x, t: self.dynamics(x=x, t=t, **kwargs)
  File "/home/flferretti/jaxsim/src/jaxsim/api/ode.py", line 69, in f
    return system_dynamics(
jax.errors.TracerBoolConversionError: Attempted boolean conversion of traced array with shape bool[1]..
The error occurred while tracing the function <lambda> at /home/flferretti/jaxsim/src/jaxsim/integrators/common.py:374 for cond. This value became a tracer due to JAX operations on these lines:

  operation a:bool[1] = eq b c
    from line /home/flferretti/jaxsim/src/jaxsim/api/ode.py:69 (f)
See https://jax.readthedocs.io/en/latest/errors.html#jax.errors.TracerBoolConversionError
```

<details>
<summary>MWE</summary>

```python
import jax.numpy as jnp
import jaxsim.api as js
import rod.builder.primitives
import rod.urdf.exporter
from jaxsim import integrators

# Create on-the-fly a ROD model of a box.
rod_model = (
    rod.builder.primitives.BoxBuilder(x=0.3, y=0.2, z=0.1, mass=1.0, name="box")
    .build_model()
    .add_link()
    .add_inertial()
    .add_visual()
    .add_collision()
    .build()
)

# Export the URDF string.
urdf_string = rod.urdf.exporter.UrdfExporter.sdf_to_urdf_string(
    sdf=rod_model, pretty=True
)

model1 = js.model.JaxSimModel.build_from_model_description(
    model_description=urdf_string,
    is_urdf=True,
)

model2 = js.model.JaxSimModel.build_from_model_description(
    model_description=urdf_string,
    is_urdf=True,
)

data1 = js.data.JaxSimModelData.build(model=model1)

data2 = js.data.JaxSimModelData.build(model=model2)

integrator1 = integrators.fixed_step.Heun2SO3.build(
    dynamics=js.ode.wrap_system_dynamics_for_integration(
        model=model1,
        data=data1,
        system_dynamics=js.ode.system_dynamics,
    ),
)

integrator2 = integrators.fixed_step.Heun2SO3.build(
    dynamics=js.ode.wrap_system_dynamics_for_integration(
        model=model2,
        data=data2,
        system_dynamics=js.ode.system_dynamics,
    ),
)

integrator_state1 = integrator1.init(x0=data2.state, t0=0, dt=1e-3)

integrator_state2 = integrator2.init(x0=data2.state, t0=0, dt=1e-3)
```
</details>

flferretti, 2024-05-22:
After some testing, I found out that the objects `data.state.physics_model` are not compatible. Moreover, when `data` is created for a model without joints, like in the example in https://github.com/ami-iit/jaxsim/issues/103#issuecomment-2058752595, the fact that the attributes `joint_positions` and `joint_velocities` are empty raises https://github.com/google/jax/blob/95c05521b457f2572b0704449c76982c85ee72b6/jax/_src/core.py#L646-L647.

Setting `hash=False` partially solves the problem, but still I cannot compare `data1` with `data2`

flferretti, 2024-05-22:
Furthermore, I believe that we should pay attention to the `dynamics` attribute of `Integrator`, which being just a Protocol, it cannot be compared with `==`

diegoferigo, 2024-05-27:
Thanks for the investigation @flferretti. Since `JaxSimModelData` contains many JAX arrays as attributes, I believe that the `__hash__` and `__eq__` methods need to be properly implemented. I'll have a look at it, I've already have some WIP work on the implementation of these methods for the `jaxsim.parser` package, I'll extend it also to `JaxSimModelData`.

closed 2024-06-04 by diegoferigo