gbionics/jaxsim #118: Package jaxsim pulled from conda-forge raises error because of conflct with new Jax update

lorycontixd, 2024-03-22:
The dependency from [Jax](https://github.com/google/jax) listed in the environment.yml file is not correct for a conda-forge installation, as it conflicts with a newer update of the Jax library.

## Error
The error is:
```ImportError: cannot import name 'config' from 'jax.config'```
from file ```src/jaxsim/__init__.py: line 9```

## How to reproduce
- The [requirements.txt](https://github.com/ami-iit/jaxsim/files/14724283/requirements.txt) file contains the output of ```conda list```:
- conda version: 23.11.0

## Possible solution
Require jax version to be < 0.4.25 in environment.yml

diegoferigo, 2024-03-22:
Hi @lorycontixd, thanks for reporting! Yes you are right, `jax >= 0.4.25` removed `jax.config`  and we used to use it in our `__init__.py`. Our `main` branch has been already fixed in #92, but we still need to make a new release and propagate it to conda-forge.

We are soon releasing a new `v0.2` with brand new APIs (follow #108 for more details). If you are starting any kind of exploration with JaxSim, I warmly recommend to install a development version from the `functional` branch. Code based on `v0.1` will no longer run on `v0.2`, and it would need major updates.

traversaro, 2024-03-22:
I guess that in the case of the `functional` branch, if you want to install jaxsim from source in a conda environment, you can use the [`environment.yml`](https://github.com/ami-iit/jaxsim/blob/functional/environment.yml) file, right @diegoferigo ?

In that the installation is basically:

~~~
git clone https://github.com/ami-iit/jaxsim -b functional
conda create -f ./jaxsim/environment.yml
conda activate jaxsim
pip install -e --no-deps ./jaxsim
~~~

diegoferigo, 2024-04-03:
Should get fixed bu https://github.com/conda-forge/jaxsim-feedstock/pull/2.

flferretti, 2024-04-03:
PR merged, closing

closed 2024-04-03 by flferretti

traversaro, 2024-04-03:
Great!