gbionics/jaxsim #139: Fixed joints related model parsing issue from URDF

berkguler, 2024-04-19:
Hi! When I build JaxSimModel from the Urdf File, it does not count the child link attached with a fixed joint. For Instance, for the following rigid body tree:

base_link (fixed joint) link_1 (revolute joint) link_2 (revolute joint) link_3 (fixed joint) link_4

When I use the jaxsim.api.JaxSimModel.build_from_model_description function, I can not see the name of link_4 in link_names(). If I did something wrong, I could provide a simple URL and code snippet to replicate the issue.

diegoferigo, 2024-04-19:
Hi @berkguler, I think what you've got is the expected behavior, the fixed joints are removed when a model is parsed, and their parent and child links get lumped together becoming a single link.

Considering your description, I'd expect 3 links (base+link1, link2, link3+link4) and the two revolute joints.

berkguler, 2024-04-21:
Thank you for your quick reply. Since I defined my tooltip as a another frame with fixed joint, I need it for the forward kinematics. Do you have any suggestions for that? In below, you can understand how I get the homogenous transformation matrix and Jacobian matrix for my corresponding frame. However, it should have derived from link_index = 8.

```
forward_kinematics(robotmodel, link_index=7,joint_positions=x)
jacobian(robotmodel, link_index=7, joint_positions=x)[:,6:])
```

diegoferigo, 2024-04-22:
Hi @berkguler, I'm not 100% sure I understood exactly what you're looking for, let me try to rephrase. You want to track in the cartesian space the location and orientation of other frames beyond those corresponding to links and joints. Therefore -as is done often in URDF- you added a new chain `parent_link -> fixed joint -> dummy link` to include the new frame of the dummy link in the kinematics.

JaxSim currently exploits a [custom parser](https://github.com/ami-iit/rod) of model descriptions supporting both SDF and URDF files. The only format supported OOTB is SDF, in fact we rely on the [sdformat](https://github.com/gazebosim/sdformat) library to convert URDF to SDF. Unfortunately, such URDF chains are removed by default by sdformat upon conversion to URDF. The first suggestion to make your model work inside JaxSim, is [altering your URDF such that fixed joint lumping is disabled](http://sdformat.org/tutorials?tut=sdformat_urdf_extensions&cat=specification&#fixed-joint-lumping). You can inspect the final SDF with `gz sdf -p /path/to/model.urdf`, if your fixed joints are preserved, the extra tags to prevent link lumping worked successfully.

As soon you can pass to JaxSim a model with a fixed joint, also JaxSim will try to remove it by lumping its parent and child links together. However, this time, JaxSim keeps track of the frame of the removed link. At the moment, this information is kept internally, but exposing all frames is a desired feature the we want to implement someday. These days I'm refactoring a similar logic in #137, as soon as it gets merged into main, you can get the transform ${}^P \mathbf{H}_F \in \text{SE}(3)$ between the frame $F$ and its parent link $P$ as follows:

```python
import numpy.typing as npt

import jaxsim.api as js
from jaxsim.parsers.kinematic_graph import KinematicGraphTransforms

def frame_relative_transform(
    model: js.model.JaxSimModel, frame_name: str
) -> tuple[str, npt.NDArray]:

    if frame_name not in model.description.get().frames_dict:
        raise ValueError(f"Frame '{frame_name}' not found in the model.")

    # Build a low-level IK.
    ik = KinematicGraphTransforms(graph=model.description.get())

    # Get the link to which the frame is rigidly attached.
    parent_link_name = (
        model.description.get().frames_dict[frame_name].parent.name
    )

    # Get the transform from the frame to the parent link.
    P_H_F = ik.transform(name=frame_name)

    return parent_link_name, P_H_F
```

Note that this IK is not the one included in the [RBDAs](https://github.com/ami-iit/jaxsim/blob/main/src/jaxsim/rbda/), it's a new one from #137 that is used in early stages only on the parsed model description. Once you have the parent-to-frame transform, you can get using `jaxsim.api.link.transform` the world transform of the parent link ${}^W \mathbf{H}_P$ and combine them as ${}^W \mathbf{H}_P {}^P \mathbf{H}_F$.

So, in short, what you observe is the expected behavior. In order to address properly your needs, we would need to expose the kinematics of frames (fully supported by SDF) beyond the kinematics of links. The method of this comment could be used as a temporary workaround.

berkguler, 2024-05-08:
Thank you for your support. `disableFixedJointLumping` fixed my issue.

diegoferigo, 2024-05-08:
> Thank you for your support. `disableFixedJointLumping` fixed my issue.

Awesome! If you are interested in computing either FK or Jacobian of additional frames of your model, please follow #148.