gbionics/jaxsim #141: Cannot run notebook examples on Colab

xela-95, 2024-04-19:
The notebooks https://colab.research.google.com/github/ami-iit/jaxsim/blob/main/examples/PD_controller.ipynb and https://colab.research.google.com/github/ami-iit/jaxsim/blob/main/examples/Parallel_computing.ipynb stops at the first cell for an `ImportError`:

```
---------------------------------------------------------------------------

ImportError                               Traceback (most recent call last)

[<ipython-input-2-1e1104358228>](https://localhost:8080/#) in <cell line: 15>()
     13 import jax
     14 import jax.numpy as jnp
---> 15 from jaxsim import logging
     16 
     17 logging.set_logging_level(logging.LoggingLevel.INFO)

1 frames

[/usr/local/lib/python3.10/dist-packages/jaxsim/__init__.py](https://localhost:8080/#) in _jnp_options()
      6     import os
      7 
----> 8     from jax.config import config
      9 
     10     # Enable by default

ImportError: cannot import name 'config' from 'jax.config' (/usr/local/lib/python3.10/dist-packages/jax/config.py)

---------------------------------------------------------------------------
NOTE: If your import is failing due to a missing package, you can
manually install dependencies using either !pip or !apt.

To view examples of installing some common dependencies, click the
"Open Examples" button below.
---------------------------------------------------------------------------
```

traversaro, 2024-04-19:
That sounds like https://github.com/ami-iit/jaxsim/issues/118 all over again. I added a `!pip list --verbose` in the notebook, and this is the output:
~~~
Package                          Version               Location                                Installer
-------------------------------- --------------------- --------------------------------------- ---------
absl-py                          1.4.0                 /usr/local/lib/python3.10/dist-packages pip
aiohttp                          3.9.5                 /usr/local/lib/python3.10/dist-packages pip
aiosignal                        1.3.1                 /usr/local/lib/python3.10/dist-packages pip
alabaster                        0.7.16                /usr/local/lib/python3.10/dist-packages pip
albumentations                   1.3.1                 /usr/local/lib/python3.10/dist-packages pip
altair                           4.2.2                 /usr/local/lib/python3.10/dist-packages pip
annotated-types                  0.6.0                 /usr/local/lib/python3.10/dist-packages pip
anyio                            3.7.1                 /usr/local/lib/python3.10/dist-packages pip
appdirs                          1.4.4                 /usr/local/lib/python3.10/dist-packages pip
argon2-cffi                      23.1.0                /usr/local/lib/python3.10/dist-packages pip
argon2-cffi-bindings             21.2.0                /usr/local/lib/python3.10/dist-packages pip
array_record                     0.5.1                 /usr/local/lib/python3.10/dist-packages pip
arviz                            0.15.1                /usr/local/lib/python3.10/dist-packages pip
astropy                          5.3.4                 /usr/local/lib/python3.10/dist-packages pip
astunparse                       1.6.3                 /usr/local/lib/python3.10/dist-packages pip
async-timeout                    4.0.3                 /usr/local/lib/python3.10/dist-packages pip
atpublic                         4.1.0                 /usr/local/lib/python3.10/dist-packages pip
attrs                            23.2.0                /usr/local/lib/python3.10/dist-packages pip
audioread                        3.0.1                 /usr/local/lib/python3.10/dist-packages pip
autograd                         1.6.2                 /usr/local/lib/python3.10/dist-packages pip
Babel                            2.14.0                /usr/local/lib/python3.10/dist-packages pip
backcall                         0.2.0                 /usr/local/lib/python3.10/dist-packages pip
beautifulsoup4                   4.12.3                /usr/local/lib/python3.10/dist-packages pip
bidict                           0.23.1                /usr/local/lib/python3.10/dist-packages pip
bigframes                        1.2.0                 /usr/local/lib/python3.10/dist-packages pip
bleach                           6.1.0                 /usr/local/lib/python3.10/dist-packages pip
blinker                          1.4                   /usr/lib/python3/dist-packages
blis                             0.7.11                /usr/local/lib/python3.10/dist-packages pip
blosc2                           2.0.0                 /usr/local/lib/python3.10/dist-packages pip
bokeh                            3.3.4                 /usr/local/lib/python3.10/dist-packages pip
bqplot                           0.12.43               /usr/local/lib/python3.10/dist-packages pip
branca                           0.7.1                 /usr/local/lib/python3.10/dist-packages pip
build                            1.2.1                 /usr/local/lib/python3.10/dist-packages pip
CacheControl                     0.14.0                /usr/local/lib/python3.10/dist-packages pip
cachetools                       5.3.3                 /usr/local/lib/python3.10/dist-packages pip
catalogue                        2.0.10                /usr/local/lib/python3.10/dist-packages pip
certifi                          2024.2.2              /usr/local/lib/python3.10/dist-packages pip
cffi                             1.16.0                /usr/local/lib/python3.10/dist-packages pip
chardet                          5.2.0                 /usr/local/lib/python3.10/dist-packages pip
charset-normalizer               3.3.2                 /usr/local/lib/python3.10/dist-packages pip
chex                             0.1.86                /usr/local/lib/python3.10/dist-packages pip
click                            8.1.7                 /usr/local/lib/python3.10/dist-packages pip
click-plugins                    1.1.1                 /usr/local/lib/python3.10/dist-packages pip
cligj                            0.7.2                 /usr/local/lib/python3.10/dist-packages pip
cloudpathlib                     0.16.0                /usr/local/lib/python3.10/dist-packages pip
cloudpickle                      2.2.1                 /usr/local/lib/python3.10/dist-packages pip
cmake                            3.27.9                /usr/local/lib/python3.10/dist-packages pip
cmdstanpy                        1.2.2                 /usr/local/lib/python3.10/dist-packages pip
colorcet                         3.1.0                 /usr/local/lib/python3.10/dist-packages pip
coloredlogs                      15.0.1                /usr/local/lib/python3.10/dist-packages pip
colorlover                       0.3.0                 /usr/local/lib/python3.10/dist-packages pip
colour                           0.1.5                 /usr/local/lib/python3.10/dist-packages pip
[... cut ...]
yellowbrick                      1.5                   /usr/local/lib/python3.10/dist-packages pip
yfinance                         0.2.38                /usr/local/lib/python3.10/dist-packages pip
zict                             3.0.0                 /usr/local/lib/python3.10/dist-packages pip
zipp                             3.18.1                /usr/local/lib/python3.10/dist-packages pip
~~~

Somehow jaxsim 0.1.* is installed instead of 0.2.0 . Probably we can remove `-q` from the `pip install` invocation to understand what's going on.

traversaro, 2024-04-19:
This is the output of `pip install -U jaxsim` on a clean colab:
~~~

Untitled2.ipynb_
Notebook unstarred

Collecting jaxsim
  Downloading jaxsim-0.1.dev389-py3-none-any.whl (94 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 94.6/94.6 kB 2.0 MB/s eta 0:00:00
Collecting coloredlogs (from jaxsim)
  Downloading coloredlogs-15.0.1-py2.py3-none-any.whl (46 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 46.0/46.0 kB 2.9 MB/s eta 0:00:00
Requirement already satisfied: jax>=0.4.1 in /usr/local/lib/python3.10/dist-packages (from jaxsim) (0.4.26)
Requirement already satisfied: jaxlib in /usr/local/lib/python3.10/dist-packages (from jaxsim) (0.4.26+cuda12.cudnn89)
Collecting jaxlie (from jaxsim)
  Downloading jaxlie-1.3.4-py3-none-any.whl (21 kB)
Collecting jax-dataclasses>=1.4.0 (from jaxsim)
  Downloading jax_dataclasses-1.6.0-py3-none-any.whl (14 kB)
Collecting pptree (from jaxsim)
  Downloading pptree-3.1.tar.gz (3.0 kB)
  Preparing metadata (setup.py) ... done
Collecting rod (from jaxsim)
  Downloading rod-0.2.0-py3-none-any.whl (37 kB)
Requirement already satisfied: typing-extensions in /usr/local/lib/python3.10/dist-packages (from jaxsim) (4.11.0)
Requirement already satisfied: ml-dtypes>=0.2.0 in /usr/local/lib/python3.10/dist-packages (from jax>=0.4.1->jaxsim) (0.2.0)
Requirement already satisfied: numpy>=1.22 in /usr/local/lib/python3.10/dist-packages (from jax>=0.4.1->jaxsim) (1.25.2)
Requirement already satisfied: opt-einsum in /usr/local/lib/python3.10/dist-packages (from jax>=0.4.1->jaxsim) (3.3.0)
Requirement already satisfied: scipy>=1.9 in /usr/local/lib/python3.10/dist-packages (from jax>=0.4.1->jaxsim) (1.11.4)
Collecting humanfriendly>=9.1 (from coloredlogs->jaxsim)
  Downloading humanfriendly-10.0-py2.py3-none-any.whl (86 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 86.8/86.8 kB 4.8 MB/s eta 0:00:00
Collecting tyro (from jaxlie->jaxsim)
  Downloading tyro-0.8.3-py3-none-any.whl (102 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 102.0/102.0 kB 6.1 MB/s eta 0:00:00
Collecting mashumaro (from rod->jaxsim)
  Downloading mashumaro-3.12-py3-none-any.whl (89 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 89.9/89.9 kB 5.3 MB/s eta 0:00:00
Requirement already satisfied: packaging in /usr/local/lib/python3.10/dist-packages (from rod->jaxsim) (24.0)
Collecting resolve-robotics-uri-py (from rod->jaxsim)
  Downloading resolve_robotics_uri_py-0.2.0-py3-none-any.whl (6.3 kB)
Collecting xmltodict (from rod->jaxsim)
  Downloading xmltodict-0.13.0-py2.py3-none-any.whl (10.0 kB)
Requirement already satisfied: docstring-parser>=0.14.1 in /usr/local/lib/python3.10/dist-packages (from tyro->jaxlie->jaxsim) (0.16)
Requirement already satisfied: rich>=11.1.0 in /usr/local/lib/python3.10/dist-packages (from tyro->jaxlie->jaxsim) (13.7.1)
Collecting shtab>=1.5.6 (from tyro->jaxlie->jaxsim)
  Downloading shtab-1.7.1-py3-none-any.whl (14 kB)
Requirement already satisfied: markdown-it-py>=2.2.0 in /usr/local/lib/python3.10/dist-packages (from rich>=11.1.0->tyro->jaxlie->jaxsim) (3.0.0)
Requirement already satisfied: pygments<3.0.0,>=2.13.0 in /usr/local/lib/python3.10/dist-packages (from rich>=11.1.0->tyro->jaxlie->jaxsim) (2.16.1)
Requirement already satisfied: mdurl~=0.1 in /usr/local/lib/python3.10/dist-packages (from markdown-it-py>=2.2.0->rich>=11.1.0->tyro->jaxlie->jaxsim) (0.1.2)
Building wheels for collected packages: pptree
  Building wheel for pptree (setup.py) ... done
  Created wheel for pptree: filename=pptree-3.1-py3-none-any.whl size=4609 sha256=17aa8f3bc51f2e1707f9cfdd59bc36a57676b0393905e565b026e3cdf23fd54b
  Stored in directory: /root/.cache/pip/wheels/9f/b6/0e/6f26eb9e6eb53ff2107a7888d72b5a6a597593956113037828
Successfully built pptree
Installing collected packages: pptree, xmltodict, shtab, resolve-robotics-uri-py, mashumaro, humanfriendly, coloredlogs, tyro, rod, jax-dataclasses, jaxlie, jaxsim
Successfully installed coloredlogs-15.0.1 humanfriendly-10.0 jax-dataclasses-1.6.0 jaxlie-1.3.4 jaxsim-0.1.dev389 mashumaro-3.12 pptree-3.1 resolve-robotics-uri-py-0.2.0 rod-0.2.0 shtab-1.7.1 tyro-0.8.3 xmltodict-0.13.
~~~

I think that the problem is that colab has Python 3.10, and jaxsim >= 0.2.0 (that fix the compatibility with jax 0.4.25) required Python 3.11 (see https://github.com/ami-iit/jaxsim/pull/67).

I guess the possible choices are:
* Support again Python 3.10 (I am not sure if we are actually using some Python >=3.11 feature or this is just a result of the metadata changed in https://github.com/ami-iit/jaxsim/pull/67)
* Change the docs to drop the references to colab (see for example https://github.com/ami-iit/jaxsim/blob/a6544fb9c1663caea2b73daebd0faa379080cf79/examples/README.md?plain=1#L17). As it is possible that Colab will support Python 3.11 soon (see https://medium.com/google-colab/colab-updated-to-python-3-10-27eb02daa162), a possible variant to this is just to add a warning for users that the tutorial will not work until colab does not update to 3.11

cc @diegoferigo @flferretti

flferretti, 2024-04-20:
The library does unofficially support python 3.10.  To make it "official", we just need to change a line in the `setup.cfg`, so it shouldn't be too hard

commit bdc8ab11 (2024-04-23): Allow backward compatibility with Python 3.10

Fixes #141