gbionics/jaxsim #146: Possibility to pass euler angles into JaxSimModelData's build

lorycontixd, 2024-04-30:
Often I find myself with the necessity of building a model's data manually, and to feed an initial rotation to this data. The static ```JaxSimModelData.build``` function of JaxSimModelData only accepts quaternions as a rotation parameter, but it would be nice to let it accept euler angles as well, as they are definitely more intuitive. 
What I suggest is to leverage the different shapes between quaternions and euler angles (quaternions's shape = 4, euler angles shape = 3) to distinguish which type of rotation has been passed to the function and perform an internal conversion.

lorycontixd, 2024-04-30:
What about other rotation formalisms which are still more intuitive than quaternions, such as angle-axis or a rotation matrix? As far as I know, these are not subject to gimbal lock. The idea is still to convert it immediately to a quaternion for internal use, but at least it's easier to build the data in the front-end side.

diegoferigo, 2024-05-02:
I'm not sure if it's worth complicating the APIs for adding another one among many other representations of $\text{SO}(3)$ elements. You don't need any third-party dependencies to convert in your downstream code your preferred representation to quaternion since JAX provides `jax.scipy.spatial.transform.Rotation`:

```python
from jax.scipy.spatial.transform import Rotation

r, p, y = 1.0, 2.0, 3.0

Q_xyzw = Rotation.from_euler("xyz", [r, p, y]).as_quat() 
```

Note that, in any case, supporting Euler angles is all but straightforward. There are way too many conventions (intrinsic/extrinsic, countless sequences, etc) and I prefer to expose unambiguous quantities like a quaternion.

closed 2024-05-02 by diegoferigo