gbionics/jaxsim #151: Address long compilation time in RBDAs

flferretti, 2024-05-15:
After an extensive debugging, I noticed that the RBDAs requires long compilation time, especially when the number of DoFs is high (> 20). I believe that the problem might be related to this method here, which uses a plain Python for-loop with `enumerate`. Moreover, the `joint_motion_subspace` uses `enum.IntEnum` which is not completely compatible with JAX.

Therefore, the idea is to find a valid JAX-compatible alternative, eventually constructing a custom enum class that automatically handles the `enum` as integers, which are compatible with JAX arrays.

diegoferigo, 2024-05-15:
> I believe that the problem might be related to this piece of code

Yes this is correct. The current structure that defines a joint type (either as a [`JointType`](https://github.com/ami-iit/jaxsim/blob/d7cfcc6b7c8aab3528a5c81e3a67f157dddade27/src/jaxsim/parsers/descriptions/joint.py#L17-L35) enum or a [`JointDescriptor`](https://github.com/ami-iit/jaxsim/blob/d7cfcc6b7c8aab3528a5c81e3a67f157dddade27/src/jaxsim/parsers/descriptions/joint.py#L38-L45) pytree) prevents structuring that list comprehension either as a `vmap` transformation or `jax.lax.scan` statement. If a model has many joints (it's not important their type), the logic of the `supported_joint_motion` function is copied over and over, and JAX has to compile a huge piece of code. It starts to be noticeable for models with more than 20/25 DoFs.

I already spent more time than I want to admit trying to find a clean and general solution to this problem. Originally we had single-axis joints (`Rx`, `Ry`, etc.) in addition to generic revolute/prismatic joints like `R` and `P` (that require the joint axis as metadata). The way to support both of them until now prevented replacing the problematic list comprehension. However, in https://github.com/ami-iit/jaxsim/pull/137, I've recently removed many of the original joint types so that the `supported_joint_motion` function could be compiled only for `R` and `P`. Note that `F` is never used.

I've performed multiple attempts by developing jit- and vmap-compatible enums, refactoring the `JointDescriptor` class, etc. None of the attempts gave me something that worked fine, and the implementation ended up being too complicated to maintain.

At this point, since this is becoming a bottleneck, I believe that resorting to a more custom / hardcoded solution and introducing few assumptions is the way to go. I will propose something in the next comment.

diegoferigo, 2024-05-15:
I propose to make the following assumption:

- JaxSim only supports revolute and prismatic joints.

This is already the case, but the previous structure allowed a simple inclusion of other joint types, and I wanted at some point to add a `FreeFloating` joint so that we could remove in our RBDAs all the if/else statement related to fixed-base / floating-base models. For the time being, I think we can live happy with only these two joint types, that are those required by probably 99% of the models out there.

---

Here below my proposal:

1. Remove the [`JointType`](https://github.com/ami-iit/jaxsim/blob/d7cfcc6b7c8aab3528a5c81e3a67f157dddade27/src/jaxsim/parsers/descriptions/joint.py#L17-L35) enum and replace it with the following class:
   ```python
    class JointType:
      Fixed: jtp.Int = jnp.array(0, dtype=int)
      Revolute: jtp.Int = jnp.array(1, dtype=int)
      Prismatic: jtp.Int = jnp.array(2, dtype=int)
   ```
2. In this way, `JointDescriptor` becomes a propert vmappable pytree since `joint_type` is a jax-compatible leaf. The same applies to [`JointGenericAxis`](https://github.com/ami-iit/jaxsim/blob/d7cfcc6b7c8aab3528a5c81e3a67f157dddade27/src/jaxsim/parsers/descriptions/joint.py#L48-L49) that is the class used to implement prismatic/revolute joints.
3. I would not change, at least in this first iteration, `JointModel.joint_types`. I think that making it non-static doesn't give us any noticeable improvement. With 2) in place, I believe that things can be vectorized elsewhere.
4. Modify [`supported_joint_motion`](https://github.com/ami-iit/jaxsim/blob/d7cfcc6b7c8aab3528a5c81e3a67f157dddade27/src/jaxsim/math/joint_model.py#L230) as follows:
   ```python
   @jax.jit
   def supported_joint_motion(
       joint_type: jtp.IntLike, joint_position: jtp.FloatLike, **kwargs
   ) -> tuple[jtp.Matrix, jtp.Array]:
   ```
   where the `axis` metadata necessary for `R` and `P` is passed as `kwarg` and unpacked in the chained `jax.lax.cond` statements that will replace the match/case (it cannot be plain if/else sequence if we want jit support over the `joint_type` integer).
5. Replace the [problematic list comprension](https://github.com/ami-iit/jaxsim/blob/d7cfcc6b7c8aab3528a5c81e3a67f157dddade27/src/jaxsim/api/kin_dyn_parameters.py#L387-L393) with something similar to the following:
   ```python
   pre_H_suc_and_S = jax.vmap(supported_joint_motion)(
      jnp.array([j.joint_type for j in self.joint_model.joint_types[1:]], dtype=int),
      jnp.array(joint_positions).astype(float),
      jnp.array([j.axis for j in self.joint_model.joint_types[1:]], dtype=float)
   )
   ```

In other works, my proposal is to only support joints that can be modeled as `JointGenericAxis`. In this way, we can vectorize over a unique pytree structure.

---

This implementation prevents the definition of other joint types where the motion subspace `S` is not a `(6, 1)` array (and the joint angle is not a scalar, in general). Again, if needed, we can think of a proper solution for that in the future.

flferretti, 2024-05-15:
In the end I kept the support for fixed joints and separated `JointType` from `JointGenericAxis`. Furthermore, I rewrote the `JointType` class so that its values get interpreted as plain integers, which simplifies a lot the computation on the JAX side. (https://github.com/ami-iit/jaxsim/commit/51ea173aa8792e4da5c5e12c188c010d7ebe6121)

Then, I added a couple of minor changes here (https://github.com/ami-iit/jaxsim/commit/0d636d9c5d7d0a1d82965406b1c9a32e9013413b) and finally used `vmap` to speed up the computation of the joint transforms and the motion subspaces in https://github.com/ami-iit/jaxsim/commit/a6ddcd2ffe7c03180189c6baa7a3becc22c17381. 

I will add some benchmarks soon

diegoferigo, 2024-05-15:
Great! I'm glad that the pattern worked fine. I'm not fully convinced on the metaclass though. I'll review the PR as soon as it's ready.

flferretti, 2024-05-15:
Benchmarks reported in https://github.com/ami-iit/jaxsim/pull/153#issuecomment-2112687460

PR #153 (2024-05-15): Improve compilation time of RBDAs for models with many DoFs
This PR drastically reduces the compilation time of RBDAs and of functions that involve joint-related calculations. In particular:
- the use of `enum.IntEnum` for the `JointType` has been reprecated in favor of a leaner class that simply instanciates subclasses for which the only value is an integer representing the joint type
- the `JointDescriptor` class has been deprecated, separating the `JointGenericAxis` from the `JointType` completely. In this way, not only we avoid a circular interdependency between attributes of `JointDescrip [... cut]
closed 2024-05-15 by flferretti