gbionics/jaxsim #159: Extend `JaxSimModelReferences` to support applying forces to frames

diegoferigo, 2024-05-22:
We already have [`JaxSimModelReferences.apply_link_forces`](https://github.com/ami-iit/jaxsim/blob/7cfca207e688ebffa9690ccdd6f225336207b4d7/src/jaxsim/api/references.py#L316-L324) to apply forces to the links. Now that frame support is finalized (#147), it would be great adding a new `JaxSimModelReferences.apply_frame_force` method that allows to apply a 6D force to a given frame. In this way, the logic can directly detect the parent link of the frame and automatically express the 6D frame force to the frame of the parent link.

flferretti, 2024-07-17:
Given a frame force ${} _F F$ expressed in inertial representation, the corresponding force on the parent link will be given by:

```math
{} _L F = {} _L X ^F {} _F F
```

in which:

```math
\begin{align}
{} _L X ^F &= \text{Adj}{} _L H ^F \\
{} _L H ^F &= ({} ^L H _F) ^\top = ({} ^L H _W {}^W H _F) ^\top
\end{align}
```

This logic can be parallelized for every considered frame.

diegoferigo, 2024-07-23:
I want to extend a bit the math since I'm reviewing #209 and there's something that might be not 100% correct.

Let's consider the following two frames:

1. A link frame $L$.
2. A generic frame $F$ rigidly attached to link $L$ and having a fixed pose defined by ${}^L \mathbf{H}_F$.

Let's now consider a 6D force $\mathbf{f}$ applied to the origin of frame $F$. The aim of the desired helper is to express such force to the link frame $L$. Since our internal buffers store velocites and forces in inertial-fixed representation, we need to also convert the frame force to inertial-fixed.

To be as generic as possible, let's consider an input 6D force expressed in frame $Y \in \left[ W, F, F[W] \right]$ that describes inertial-fixed, body-fixed, and mixed representations.

What we need to do is the following:

```math
{}_L \mathbf{f} = {}_L \mathbf{X}^Y {}_Y \mathbf{f}
```

where ${}_L \mathbf{X}^Y = {}^Y \mathbf{X}_L^\top = {}^L \mathbf{X}_Y^{-\top} $.

flferretti, 2024-07-24:
Thanks a lot @diegoferigo! I've updated my comment accordingly

PR #209 (2024-07-24): Add the possibility to set forces to frames
The implementation extends `JaxSimModelReferences` with a `apply_frame_forces` method that first converts the forces to inertial representation, then converts them to the parent link forces and finally calls `apply_link_forces` to return the object with the updated link forces. 
The conversion logic can be found at https://github.com/ami-iit/jaxsim/issues/159#issuecomment-2233371413.

Closes #159 

----
📚 Documentation preview 📚: https://jaxsim--209.org.readthedocs.build//209/
closed 2024-07-26 by diegoferigo