gbionics/jaxsim #175: Extend the documentation with an textbook-like introduction

markusheimerl, 2024-06-13:
see https://mujoco.readthedocs.io/en/stable/computation/index.html for reference.

diegoferigo, 2024-06-13:
Hi @markusheimerl, thanks for the suggestion. We are aware that the documentation is currently lacking, so far we focused mainly on functionality. Many things have changed in the past few months, but now we reached APIs that are much more stable.

We are considering both to extend the documentation and provide a paper / technical report. Timing is not yet clear though.

markusheimerl, 2024-06-13:
Hello @diegoferigo, thanks for the prompt update. Looking forward to developments of jaxsim 👍🏻

diegoferigo, 2024-06-28:
Waiting for more details that go beyond the docstrings of our code, for the time being I would probably redirect interested readers to Chapter 2 (_Robot modelling_), Chapter 7 (_Contact-aware multibody dynamics_) and Chapter 8 (_Scaling rigid-body simulations_) of my Ph.D. thesis[^1]. It describes the considered dynamics, the soft contact model, and provides the pseudocode of the rigid body dynamics algorithms implemented in JaxSim.

Many software-engineering things have changed since, but the modelling is pretty much the same, and at least readers can find a good reference of notation and considered dynamics.

[^1]: **Simulation Architectures for Reinforcement Learning applied to Robotics**, 2022, [URL](https://github.com/diegoferigo/phd-thesis).

diegoferigo, 2024-10-17:
We just merged in #265 and #261 new notebooks with few interesting use-cases. They might be worth a look for a primer on the considered use-cases.

closed 2025-11-06 by markusheimerl