gbionics/jaxsim #176: Creating a minimal example to tackle 'ValueError: The truth value of an array with more than one element is ambiguous' when building JaxSimModelData instance

lorycontixd, 2024-06-13:
This issue starts from [my robot implementation](https://github.com/ami-iit/element_lower-leg-morphology-optimization/blob/main/src/passive_walker/3-Learner/learner_grid.py), where building a JaxSimModelData instance inside a loop raises the following error:

> Traceback (most recent call last):
  File "/mnt/c/Users/lconti/Documents/programming/dev/element_lower-leg-morphology-optimization/src/passive_walker/3-Learner/learner_grid.py", line 661, in <module>
    main()
  File "/mnt/c/Users/lconti/Documents/programming/dev/element_lower-leg-morphology-optimization/src/passive_walker/3-Learner/learner_grid.py", line 619, in main
    soft_contacts_params=js.contact.estimate_good_soft_contacts_parameters(
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 4, in __eq__
  File "/home/lorenzoconti/miniforge3/envs/pw_learning/lib/python3.12/site-packages/jax/_src/array.py", line 258, in __bool__
    core.check_bool_conversion(self)
  File "/home/lorenzoconti/miniforge3/envs/pw_learning/lib/python3.12/site-packages/jax/_src/core.py", line 654, in check_bool_conversion
    raise ValueError("The truth value of an array with more than one element"
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()

As of now, I tried recreating a minimal working example that attempts to recreate the problem, but this example runs correctly.

<details>
<summary>Code</summary>

Code is found [here](https://github.com/ami-iit/element_lower-leg-morphology-optimization/commit/0f110619bda1ba1e7503e22ab8e94a5978c5d438).
</details>

<details>
<summary>Conda env</summary>

[requirements.txt](https://github.com/user-attachments/files/15821007/requirements.txt)

</details>

Next step is to add more lines from the original file to try to break the code with the traceback found above.

CC: @flferretti

flferretti, 2024-06-13:
Hi @lorycontixd, can you please post the result from `diff -u  working_version.py non_working_version.py`? Thanks a lot

lorycontixd, 2024-06-13:
> Hi @lorycontixd, can you please post the result from `diff -u working_version.py non_working_version.py`? Thanks a lot

I updated the original comment with the output of the diff. I have a feeling the code block has a length limit so I uploaded a file for it.

lorycontixd, 2024-06-13:
Added joint initial positions & link forces

- Code [here](https://github.com/ami-iit/element_lower-leg-morphology-optimization/pull/8/commits/e3fdb1794d73ab5fb0337a52daa1551d17afddef)
- Diff [here](https://github.com/user-attachments/files/15821427/diff.txt)

**Result**: still no breaking of the code

diegoferigo, 2024-06-13:
@lorycontixd can you please add snippets of MRE in code boxes wrapped by a details HTML environment? The repositories you linked are mostly private and people reading this issue have no access.

lorycontixd, 2024-06-13:
> @lorycontixd can you please add snippets of MRE in code boxes wrapped by a details HTML environment? The repositories you linked are mostly private and people reading this issue have no access.

Sure, do code boxes have a maximum limit? Because the code is 400+ lines and most of the times I have difficulties wrapping code in the appropriate boxes.

diegoferigo, 2024-06-13:
There are no significant limits AFAIK. However, in order to help other people providing you support, the shorter the MRE is, the better.

lorycontixd, 2024-06-13:
I updated the original comment to link state that the code makes use of #156 @diegoferigo

flferretti, 2024-06-14:
Closed by #177

closed 2024-06-14 by flferretti