gbionics/jaxsim #190: Consider running in CI a type checker like `mypy`

diegoferigo, 2024-06-28:
The `jaxsim` package is widely type hinted, however at the moment it's only useful while developing for IDE suggestions. It would be great if, slowly, we can start having `mypy` running at least on the `jaxsim.api` package.

The starting point is not that easy. I get ~200 errors.

- https://mypy.readthedocs.io/en/stable/index.html

flferretti, 2024-08-26:
Some people are working to implement a static type checker in [`ruff`](https://github.com/astral-sh/ruff) as documented in https://github.com/astral-sh/ruff/issues/3893. The main progresses can be tracked in the `red knot` https://github.com/astral-sh/ruff/tree/main/crates/red_knot or at https://github.com/astral-sh/ruff/issues/11653.

After trying some alternatives, I've found [`pyright`](https://github.com/microsoft/pyright) to be easier to understand and use than [`mypy`](https://github.com/python/mypy). Nevertheless, a faster and even clearer alternative could be [`pylyzer`](https://github.com/mtshiba/pylyzer), whose logging output is similar to the one reported by [`ruff`](https://github.com/astral-sh/ruff). The only drawback is that the installation requires [`cargo`](https://doc.rust-lang.org/cargo/getting-started/installation.html) as the version on PyPI cannot install [`ERG`](https://erg-lang.org)

diegoferigo, 2024-08-26:
Awesome, if ruff gains support of a static type checker, it'd be great exploiting it. We don't need many features, something quite basic could work at least in the beginning.