gbionics/jaxsim #195: Consider adding `pre-commit.ci` instead for style checking in CI

flferretti, 2024-07-04:
By using [`pre-commit.ci`](https://pre-commit.ci/), we could eliminate the file related to the `style` CI configuration, e.g. [.github/workflows/style.yml](https://github.com/ami-iit/jaxsim/blob/main/.github/workflows/style.yml) and substitute it with something that leverages the already present `.pre-commit-config.yml` and the configuration from the `pyproject.toml`.

This would allow us to add additional checks (see https://github.com/pre-commit/pre-commit-hooks), especially the `check-added-large-files`, which would prevent the user to commit the `pixi.lock` pulled from `lfs`, and the others checks already present in the pre-commit configuration.

diegoferigo, 2024-07-05:
Running the pre-commit check also on CI would be great! About the style checks (isort and black), I'd would just make sure about the following:

- We can constrain the version of black as we currently do. In the past, new major releases (once per year, usually around January) introduced breaking changes. I prefer to bump manually the black pinning.
- Right now the CI provides a nicely formatted diff in case there is a style mismatch. Before moving to ruff, let's be sure to get something comparable.

flferretti, 2024-07-05:
The formatting packages are already constrained in the `.pre-commit-config.yml` for both `black`:

https://github.com/ami-iit/jaxsim/blob/5b96fab839b84985030d78ca363a4d9c97e1d8d5/.pre-commit-config.yaml#L12

and `isort`:

https://github.com/ami-iit/jaxsim/blob/5b96fab839b84985030d78ca363a4d9c97e1d8d5/.pre-commit-config.yaml#L18

Do you prefer to have an error raised with the diff or just let the CI format the commited files?

diegoferigo, 2024-07-05:
Nope the pinning looks good. Can we match it with what specified in `pyproject.toml`? Considering the update policies, I guess that pinning the major versions would suffice.

flferretti, 2024-07-05:
The pre-commit configuration needs an exact tag or commit SHA. Can we stick with the current configuration since it matches:

```toml
# From pyproject.toml in #193 
"black[jupyter] ~= 24.0"
```
?

diegoferigo, 2024-07-05:
Yes that sounds good, I thought it could handle less strict versioning.

closed 2024-07-11 by flferretti

flferretti, 2024-10-25:
FYI @giulero

PR #197 (2024-07-11): Enable `pre-commit.ci` for style checking
This PR removes the current style checking from GitHub Actions and moves it to `pre-commit.ci`, this will allow for additional checking.

Closes #195 

----
📚 Documentation preview 📚: https://jaxsim--197.org.readthedocs.build//197/