gbionics/jaxsim #196: PlaneTerrain eq operator raising TracerBoolConversionError when simulating with Optuna

lorycontixd, 2024-07-05:
Following the __Minimal Reproducible Example__ below, I'd like to report an error raised when simulating any model using Optuna (Found below in the traceback section).
In fact, running a simulation using `RungeKutta4SO3` integrator outside of an Optuna trial, the simulation terminates successfully. However, when the simulation is run inside Optuna's trial (even on a single core), the integrator fails to initialize due to the following traceback:

<details><summary>Traceback</summary>
<p>

```bash
Traceback (most recent call last):
  File "/home/lorenzoconti/miniforge3/envs/jaxsim/lib/python3.11/site-packages/optuna/study/_optimize.py", line 196, in _run_trial
    value_or_values = func(trial)
                      ^^^^^^^^^^^
  File "/mnt/c/Users/lconti/Documents/programming/dev/element_lower-leg-morphology-optimization/src/passive_walker/Learner/integrator_test.py", line 95, in obj
    finaldata = simulate(model, data)
                ^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/lconti/Documents/programming/dev/element_lower-leg-morphology-optimization/src/passive_walker/Learner/integrator_test.py", line 71, in simulate
    integrator_state = integrator.init(x0=data.state, t0=t0, dt=dt)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/lconti/Documents/programming/dev/ami-forks/jaxsim/src/jaxsim/integrators/common.py", line 165, in init
    _ = integrator(x0, t0, dt, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/lconti/Documents/programming/dev/ami-forks/jaxsim/src/jaxsim/integrators/common.py", line 282, in __call__
    z = self._compute_next_state(x0=x0, t0=t0, dt=dt, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/lconti/Documents/programming/dev/ami-forks/jaxsim/src/jaxsim/integrators/common.py", line 416, in _compute_next_state
    K, _ = jax.lax.scan(
           ^^^^^^^^^^^^^
  File "/mnt/c/Users/lconti/Documents/programming/dev/ami-forks/jaxsim/src/jaxsim/integrators/common.py", line 402, in scan_body
    ki = jax.lax.cond(
         ^^^^^^^^^^^^^
  File "/mnt/c/Users/lconti/Documents/programming/dev/ami-forks/jaxsim/src/jaxsim/integrators/common.py", line 373, in <lambda>
    get_ẋ0 = lambda: self.params.get("dxdt0", f(x0, t0)[0])
                                               ^^^^^^^^^
  File "/mnt/c/Users/lconti/Documents/programming/dev/ami-forks/jaxsim/src/jaxsim/integrators/common.py", line 364, in <lambda>
    f = lambda x, t: self.dynamics(x=x, t=t, **kwargs)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/lconti/Documents/programming/dev/ami-forks/jaxsim/src/jaxsim/api/ode.py", line 69, in f
    return system_dynamics(
           ^^^^^^^^^^^^^^^
  File "/mnt/c/Users/lconti/Documents/programming/dev/ami-forks/jaxsim/src/jaxsim/terrain/terrain.py", line 151, in __eq__
    if not (
jax.errors.TracerBoolConversionError: Attempted boolean conversion of traced array with shape bool[]..
The error occurred while tracing the function <lambda> at /mnt/c/Users/lconti/Documents/programming/dev/ami-forks/jaxsim/src/jaxsim/integrators/common.py:373 for cond.
```

</p>
</details> 

<details><summary>Minimal Reproducible Example</summary>
<p>

```python
import jax.numpy as jnp
import jaxsim
import jaxsim.api as js
import numpy as np
import optuna
import rod
from rod.builder.primitives import BoxBuilder, PrimitiveBuilder
from rod.urdf.exporter import UrdfExporter

def build_model():
    rootbuilder = BoxBuilder(
        name="Box model",
        mass=1.0,
        x=0.3,
        y=0.3,
        z=0.3,
    )
    root = (
        rootbuilder.build_link(
            name=rootbuilder.name,
            pose=PrimitiveBuilder.build_pose(
                relative_to="__model__",
            ),
        )
        .add_inertial()
        .add_visual()
        .add_collision()
        .build()
    )
    rodmodel = rod.Model(name="passive_walker", canonical_link=root.name, link=[root])
    rod_sdf = rod.Sdf(version="1.7", model=rodmodel)
    urdf_string = UrdfExporter.sdf_to_urdf_string(
        rod_sdf, pretty=True, gazebo_preserve_fixed_joints=True
    )
    terrain = jaxsim.terrain.PlaneTerrain.build(plane_normal=[0, 0, 1])
    jaxsim_model = js.model.JaxSimModel.build_from_model_description(
        model_description=urdf_string,
        model_name=rodmodel.name,
        terrain=terrain,
        is_urdf=True,
    )
    return jaxsim_model

def simulate(
    model, data, integrator: jaxsim.integrators.Integrator = None, tf=3.0, dt=0.001
):
    model: js.model.JaxSimModel
    data: js.data.JaxSimModelData

    data.reset_base_position(
        jnp.array(
            [
                0,
                0,
                np.random.uniform(0.5, 1),
            ]
        )
    )

    t0 = 0.0
    if integrator is None:
        integrator = jaxsim.integrators.fixed_step.RungeKutta4SO3.build(
            dynamics=js.ode.wrap_system_dynamics_for_integration(
                model=model,
                data=data,
                system_dynamics=js.ode.system_dynamics,
            ),
        )
    integrator_state = integrator.init(x0=data.state, t0=t0, dt=dt)

    while data.time() < tf:
        data, integrator_state = js.model.step(
            dt=dt,
            model=model,
            data=data,
            integrator=integrator,
            integrator_state=integrator_state,
            link_forces=None,
        )

    return data

def obj(trial):
    model = build_model()
    contact_params = jaxsim.rbda.contacts.soft.SoftContactsParams.build()
    data = js.data.JaxSimModelData.build(
        model=model,
        velocity_representation=jaxsim.VelRepr.Mixed,
        contacts_params=contact_params,
    )

    finaldata = simulate(model, data)
    print(finaldata)

def main():
    print("First simulate without optuna")
    model = build_model()
    contact_params = jaxsim.rbda.contacts.soft.SoftContactsParams.build()
    data = js.data.JaxSimModelData.build(
        model=model,
        velocity_representation=jaxsim.VelRepr.Mixed,
        contacts_params=contact_params,
    )
    _ = simulate(model, data)

    print("\nThen simulate with Optuna")
    study = optuna.create_study()
    study.optimize(obj, n_trials=100)

    print("Simulation done")
    print("Best value: ", study.best_value)
    print("Best params: ", study.best_params)

if __name__ == "__main__":
    main()

```

</p>
</details> 

<details><summary>Environment info</summary>
<p>

From ` conda list | grep 'jaxsim\|rod\|jax\|jaxlie'`:
```bash
# packages in environment at /home/lorenzoconti/miniforge3/envs/jaxsim:
jax                       0.4.27             pyhd8ed1ab_0    conda-forge
jax-dataclasses           1.6.0              pyhd8ed1ab_0    conda-forge
jaxlib                    0.4.23          cuda120py311h5a49573_202    conda-forge
jaxlie                    1.4.1              pyhd8ed1ab_0    conda-forge
jaxsim                    0.2.1.dev268             pypi_0    pypi
rod                       0.3.0              pyh0f4e4df_1    conda-forge
```

</p>
</details> 

I tried multiple solutions, such as bitwise operator `a | b` and `jax.logical_or()`, but without any success.

diegoferigo, 2024-07-05:
Thanks @lorycontixd for reporting. It's not clear to me what is your current JaxSim version, I don't see any `0.2.1.dev268` [in PyPI](https://pypi.org/project/jaxsim/#history).

lorycontixd, 2024-07-05:
I'm not sure as well, but I confirm I'm aligned with the current version of the main branch.
In fact, below are the top three records of my `git log`:

<details><summary>Jaxsim Git Log</summary>
<p>

```
commit 5b96fab839b84985030d78ca363a4d9c97e1d8d5 (HEAD -> main, upstream/main)
Merge: 65ed0c9 8906c9c
Author: Filippo Luca Ferretti <102977828+flferretti@users.noreply.github.com>
Date:   Wed Jul 3 18:46:06 2024 +0200

    Merge pull request #187 from ami-iit/feature/docs_examples

    Add example notebooks in documentation

commit 8906c9c7cb962fbc3324b167e866b09186f3fa8b
Author: Filippo Luca Ferretti <filippo.luca.ferretti@campus.tu-berlin.de>
Date:   Wed Jul 3 18:06:05 2024 +0200

    Address reviews

    Co-authored-by: Diego Ferigo <diego.ferigo@iit.it>

commit fed4ede8c6a67dab055fd19b2ce0bc73a64a307c
Author: Filippo Luca Ferretti <filippo.luca.ferretti@campus.tu-berlin.de>
Date:   Tue Jul 2 18:25:09 2024 +0200

    Increase cell execution timeout
```

</p>
</details>

lorycontixd, 2024-07-05:
⚠️ Jaxsim Version Update
My bad, I successfully updated to latest PyPI version `jaxsim-0.3.1.dev113`. However, the problem still persists.

<details><summary>Evironment Info</summary>
<p>

```bash
# packages in environment at /home/lorenzoconti/miniforge3/envs/jaxsim:
jax                       0.4.27             pyhd8ed1ab_0    conda-forge
jax-dataclasses           1.6.0              pyhd8ed1ab_0    conda-forge
jaxlib                    0.4.23          cuda120py311h5a49573_202    conda-forge
jaxlie                    1.4.1              pyhd8ed1ab_0    conda-forge
jaxsim                    0.3.1.dev113             pypi_0    pypi
rod                       0.3.0              pyh0f4e4df_1    conda-forge
```

</p>
</details>

diegoferigo, 2024-07-05:
Yep I reproduced the problem on my system. I added recently the new hash function in https://github.com/ami-iit/jaxsim/commit/e86255e1e341a6b0209c8f1f316b4ab58b840d16. The solution seems to replace `jnp` with `np`, probably this function is called inside a `jax.jit` context when run it optuna (not sure why) and all `jnp.array` become traced. You can fix it for now locally so that you can proceed locally, I'll prepare soon a PR.

Thanks for reporting!

lorycontixd, 2024-07-05:
Great! Thanks a lot

PR #198 (2024-07-05): Fix `PlaneTerrain.__eq__` when called in jit context
Closes #196.

cc @lorycontixd

----
📚 Documentation preview 📚: https://jaxsim--198.org.readthedocs.build//198/
closed 2024-07-05 by diegoferigo