gbionics/jaxsim #199: Add a scheduled workflow to automatically update the `pixi.lock` file

diegoferigo, 2024-07-05:
It would be nice to have a scheduled workflow (let's say, once per month) that pushes re-creates from scratch a new `pixi.lock` file and run the tests on the new dependencies. If successful, a new PR can be opened so that the new file can be pushed to LFS.

This would allow to maintain a rolling-release set of working dependencies attached to the git repository.

flferretti, 2024-07-05:
A possible idea could be to modify https://pixi.sh/dev/advanced/updates_github_actions/#how-to-use integrating LFS and logging some diff in the PR description.

diegoferigo, 2024-07-05:
Yep, we already discussed this in https://github.com/ami-iit/jaxsim/pull/189#issuecomment-2194525004.

closed 2024-08-21 by flferretti