gbionics/jaxsim #203: `test_simulation::test_box_with_zero_gravity` generates NaNs in `VelRepr.Inertial` with 10N force

flferretti, 2024-07-11:
When defining the applied forces in https://github.com/ami-iit/jaxsim/blob/2056e707805d7281c48b19574bf1d588e3c88e93/tests/test_simulations.py#L118-L122

if I scale up the sampling range from $\verb |[-1, 1]| N$ to $\verb |[-10, 10]| N$, the base position result in an array of NaNs only in inertial velocity representation. The only solution in to avoid NaNs is to reduce the integration step size, yet this should not be necessary.

diegoferigo, 2024-07-11:
Luckily JAX provides great utilities to debug NaNs. You can easily debug this failure by running the following:

```bash
JAX_DEBUG_NANS=True pytest -vv -s tests/test_simulations.py -k "test_box_with_zero_gravity[inertial]"
```

It will fail in the integration step.

---

This being said, I gave a quick look to the test and it seems to me that, in `VelRepr.Inertial`, the body receives not only linear forces (like in `Body` and `Mixed`) but also torques. In fact, the pure linear force `L_f` (that I'd write with our notation as ${}_W \boldsymbol{f}_L$), when is inserted in the `JaxSimModelReferences` and applied to the link frame `L`, it is equivalent to the sum of a linear force and a torque in `Mixed|Body`. So the two systems are different in the sense that in `Inertial`, the body receives also a torque component.

With the large $\Delta t = 0.010$ used in this test, I suspect that such torque would produce large angular accelerations that lead to the emergence of simulated NaNs.

I fixed the test in #204.

So eventually there was no problem in JaxSim. The problem was how the test was designed. cc @DanielePucci @xela-95

flferretti, 2024-07-12:
Alright, it's good to know that there was no problem in JaxSim! Thanks a lot for the PR

diegoferigo, 2024-07-12:
> Alright, it's good to know that there was no problem in JaxSim! Thanks a lot for the PR

Yep it was a concern I had, I didn't see any reason why the integrator would diverge in `VelRepr.Inertial` and not in the other representations. The only sources of differences would be the conversions among the representations used to pre-process or post-process the data of our RBDA, but those are thoroughly tested against iDynTree and also with self-consistency against other functions of JaxSim.

I'm relieved that the problem was a methodological mistake in the test.

PR #204 (2024-07-12): Fix `test_simulations::test_box_with_zero_gravity`
Closes #203.

----
📚 Documentation preview 📚: https://jaxsim--204.org.readthedocs.build//204/
closed 2024-07-12 by diegoferigo