gbionics/jaxsim #208: Add frame Jacobian derivative

xela-95, 2024-07-17:
In order to compute the acceleration of a frame it would be useful that Jaxsim exposes the Jacobian derivative $\dot J$ for frames, as it is already done for [links](https://github.com/ami-iit/jaxsim/blob/ba9b8da97f72fd51ddc2a1c771991f89c9a8a9a4/src/jaxsim/api/link.py#L389).

xela-95, 2024-07-17:
@diegoferigo @flferretti please double check the math above when you have time :)

flferretti, 2024-07-17:
Thanks a lot for your work @xela-95! It looks good to me overall, I just have a couple doubts.

> ${} ^O X _W = I _4$

I guess it should be ${} ^O X _W = I _6$

> {} ^W \mathbf{v} _{W, W \times} &= 0 _6

It should probably be a ${} ^W \mathbf{v} _{W, W \times} &= 0 _{6 \times 6}$, but I guess you wanted to indicate exactly that.

> ```math 
> {} ^F H _W \begin{bmatrix} I _3 & {} ^F o _W \\ 0 _{3 \times 1} & 1 \end{bmatrix}
> ```

I'm not sure whether an equal sign is missing and I guess that the zero should be a $0 _{1 \times 3}$ (same goes for the mixed):

```math 
{} ^F H _W 
\color{red}{=}
\begin{bmatrix} I _3 & {} ^F o _W \\ 0 _{1 \times 3} & 1 \end{bmatrix}
```

xela-95, 2024-07-17:
Thanks a lot @flferretti! I updated the math accordingly.

diegoferigo, 2024-07-17:
@xela-95 I double checked your calculations and they look correct also to me. As I suggested last time, I recommend to compute for body-fixed and mixed representations the transforms ${}^W \mathbf{H}\_F$ and ${}^W \mathbf{H}\_{F[W]}$ first, and then invert them with `jaxsim.math.Transform.inverse`. It's quite common in the implementation to forget the rotation term necessary to compute ${}^F \mathbf{p}_W$, it's easier to just get ${}^W \mathbf{p}_F$ from FK and invert the transform with the helper (that is better wrt `jnp.linalg.inv` because it does not invert anything, it just relies on transposition and sign inversion).

xela-95, 2024-07-18:
Since the test against AD was passing for inertial and body representation and not for mixed, I double checked the formulation to compute ${} ^W \mathbf{v} _{W, F[W]}$.

Since the definition of right-trivialized velocity is:

```math
{} ^A \mathbf{v} _{A, B} = 
\begin{bmatrix}
{} ^A \dot o _B - {} ^A \dot R _B {} ^A R _B ^T  {} ^A o _B \\
{} ^A \dot R _B {} ^A R _B ^T 
\end{bmatrix}
```

if instead of $B$ we use $B[A]$ we get:
```math
{} ^A \mathbf{v} _{A, B[A]} = 
\begin{bmatrix}
{} ^A \dot o _B - {} ^A \dot R _A {} ^A R _A ^T  {} ^A o _B \\
{} ^A \dot R _A {} ^A R _A ^T 
\end{bmatrix} = 
\begin{bmatrix}
{} ^A \dot o _B \\
0 _{3 \times 1}
\end{bmatrix}
```

hence this kind of velocity has the linear part of the mixed velocity ${} ^{B[A]} \mathbf{v} _{A,B}$ and zero as angular component.

diegoferigo, 2024-07-19:
In view of https://github.com/ami-iit/jaxsim/pull/210#discussion_r1684125204, would it be helpful to summarize publicly the theory we developed for the computation of the frame Jacobian? And I mean, the formulation that includes the three components of the jacobian derivative.

xela-95, 2024-07-19:
> In view of [#210 (comment)](https://github.com/ami-iit/jaxsim/pull/210#discussion_r1684125204), would it be helpful to summarize publicly the theory we developed for the computation of the frame Jacobian? And I mean, the formulation that includes the three components of the jacobian derivative.

Yep absolutely, I'm doing it right now

diegoferigo, 2024-07-19:
I just remembered that I wrote the approach in #169 since it was necessary also for the computation of $\dot{J}$ of the link frames. Maybe you can take inspiration from that and just write that are the differences in the case of frames that are rigidly attached to links.

In other words, we compute the derivative of the jacobian of a link frame ${}^O \dot{J}\_{W,L/B}$ using a RBDA operating on the kinematic tree that computes the full jacobian derivative ${}^B \dot{J}\_{W, \_/B}$. The three-derivative formulation is used also there for obtaining the real link jacobian in the three supported velocity representations. A similar processing is needed for frames rigidly attached to links, as this issue and the corresponding PR implement.

xela-95, 2024-07-19:
Just updated the mathematical description in https://github.com/ami-iit/jaxsim/issues/208#issuecomment-2233191571

PR #210 (2024-07-19): Add function to compute the Jacobian derivative of additional frames rigidly attached to links
Closes #208 

----
📚 Documentation preview 📚: https://jaxsim--210.org.readthedocs.build//210/
closed 2024-07-19 by diegoferigo