gbionics/jaxsim #259: Explore XLA flags for enhancing GPU performance

diegoferigo, 2024-10-08:
I want to open a container issue to discuss XLA flags and whether their usage affect performance on JaxSim-based applications. Here below some resources I found (feel free to extend):

- JAX documentation: https://jax.readthedocs.io/en/latest/gpu_performance_tips.html
- JAX-Toolbok: https://github.com/NVIDIA/JAX-Toolbox?tab=readme-ov-file#environment-variables

If any of our users has explored the usage of these flags and has bits of knowledge to share, this is the right place.

flferretti, 2024-12-05:
Using:

- `XLA_FLAGS="--xla_gpu_enable_latency_hiding_scheduler=true --xla_gpu_enable_triton_gemm=false"`
- `CUDA_DEVICE_MAX_CONNECTIONS=1` 
- `NCCL_NVLS_ENABLE=0`

with the benchmark suite on **NVIDIA GeForce RTX 4060**:

```bash
XLA_FLAGS="--xla_gpu_enable_latency_hiding_scheduler=true --xla_gpu_enable_triton_gemm=false" \
CUDA_DEVICE_MAX_CONNECTIONS=1 \
NCCL_NVLS_ENABLE=0 \
pixi run benchmark --batch-size 512
```
we do get some improvements in terms of computational speed.

```diff
#---------------------------------------------------------------------------------------------------- benchmark: 18 tests -----------------------------------------------------------------------------------------------------
Name (time in ms)                                      Min                   Max                  Mean             StdDev                Median                IQR            Outliers       OPS            Rounds  Iterations
#------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
-test_forward_kinematics[1]                          1.8009 (1.0)          4.3127 (1.06)         2.9470 (1.25)      0.9253 (12.30)        3.3940 (1.51)      1.8846 (22.37)       143;0  339.3328 (0.80)        256           1
+test_forward_kinematics[1]                          1.1569 (1.00)         2.6362 (1.51)         1.2331 (1.0)       0.1248 (2.57)         1.2024 (1.0)       0.0423 (1.00)        16;36  810.9639 (1.0)         640           1

-test_free_floating_jacobian[1]                      2.0132 (1.12)         4.2390 (1.04)         3.2654 (1.38)      0.6851 (9.10)         3.5514 (1.58)      1.4286 (16.96)       128;0  306.2451 (0.72)        372           1
+test_free_floating_jacobian[1]                      1.1515 (1.0)          1.7512 (1.0)          1.2985 (1.05)      0.0486 (1.0)          1.2853 (1.07)      0.0422 (1.0)        154;52  770.1047 (0.95)        603           1

-test_free_floating_jacobian_derivative[1]           2.1137 (1.17)         4.3813 (1.07)         2.3581 (1.0)       0.3296 (4.38)         2.2545 (1.0)       0.1271 (1.51)        22;32  424.0673 (1.0)         227           1
+test_free_floating_jacobian_derivative[1]           1.6339 (1.42)         2.3664 (1.35)         1.8074 (1.47)      0.0903 (1.86)         1.7874 (1.49)      0.0985 (2.33)        87;35  553.2864 (0.68)        506           1

-test_soft_contact_model[1]                          2.4866 (1.38)         4.0777 (1.0)          2.7673 (1.17)      0.2771 (3.68)         2.6664 (1.18)      0.2139 (2.54)        53;51  361.3670 (0.85)        297           1
+test_soft_contact_model[1]                          2.4945 (2.17)         3.8433 (2.19)         2.8284 (2.29)      0.2661 (5.48)         2.7190 (2.26)      0.3033 (7.18)         98;3  353.5630 (0.44)        350           1

-test_forward_dynamics_aba[1]                        2.6070 (1.45)        11.1712 (2.74)         5.0564 (2.14)      0.9521 (12.65)        5.2323 (2.32)      0.6709 (7.96)        33;29  197.7706 (0.47)        220           1
+test_forward_dynamics_aba[1]                        1.8135 (1.57)         2.6578 (1.52)         2.1470 (1.74)      0.1168 (2.40)         2.1582 (1.79)      0.0756 (1.79)        50;36  465.7594 (0.57)        268           1

-test_free_floating_jacobian[512]                    4.8500 (2.69)         6.9356 (1.70)         5.4233 (2.30)      0.6499 (8.64)         5.0692 (2.25)      1.1290 (13.40)        39;0  184.3902 (0.43)        154           1
+test_free_floating_jacobian[512]                    4.3450 (3.77)         5.5881 (3.19)         4.7580 (3.86)      0.4943 (10.18)        4.4420 (3.69)      1.0611 (25.14)        54;0  210.1728 (0.26)        173           1

-test_free_floating_mass_matrix[1]                   5.0083 (2.78)        11.3466 (2.78)         5.9254 (2.51)      0.9890 (13.14)        5.5988 (2.48)      0.7360 (8.74)         12;9  168.7660 (0.40)        122           1
+test_free_floating_mass_matrix[1]                   3.5271 (3.06)         5.2365 (2.99)         4.2388 (3.44)      0.5821 (11.99)        3.9360 (3.27)      1.2978 (30.74)        58;0  235.9166 (0.29)        184           1

-test_free_floating_bias_forces[1]                   7.1066 (3.95)         8.6387 (2.12)         7.6323 (3.24)      0.3866 (5.14)         7.5881 (3.37)      0.5891 (6.99)         39;0  131.0223 (0.31)        124           1
+test_free_floating_bias_forces[1]                   4.8151 (4.18)         7.8293 (4.47)         5.7977 (4.70)      0.4440 (9.14)         5.9511 (4.95)      0.3710 (8.79)        45;29  172.4832 (0.21)        190           1

-test_free_floating_mass_matrix[512]                 8.2956 (4.61)        12.0640 (2.96)         9.1814 (3.89)      0.8170 (10.86)        8.8992 (3.95)      0.5695 (6.76)        16;10  108.9164 (0.26)         85           1
+test_free_floating_mass_matrix[512]                 7.2602 (6.31)         8.9097 (5.09)         7.5981 (6.16)      0.1468 (3.02)         7.5955 (6.32)      0.0791 (1.87)         10;9  131.6123 (0.16)        128           1

-test_free_floating_jacobian_derivative[512]        11.0173 (6.12)        11.4923 (2.82)        11.1830 (4.74)      0.0753 (1.0)         11.1652 (4.95)      0.0843 (1.0)          18;4   89.4216 (0.21)         92           1
[... cut ...]
+test_rigid_contact_model[512]                   1,574.7430 (>1000.0)  1,587.2042 (906.37)   1,579.1300 (>1000.0)   4.7573 (97.96)    1,577.7808 (>1000.0)   4.4303 (104.95)        1;0    0.6333 (0.00)          5           1
#------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
```

FYI @traversaro @xela-95 @diegoferigo @CarlottaSartore

flferretti, 2024-12-13:
Closed as completed