gbionics/jaxsim #292: pixi.lock is not up-to-date with the project

traversaro, 2024-11-19:
I just run:

~~~
traversaro@IITBMP014LW012:~$ git lfs install
Git LFS initialized.
traversaro@IITBMP014LW012:~$ git clone https://github.com/ami-iit/jaxsim
Cloning into 'jaxsim'...
remote: Enumerating objects: 10050, done.
remote: Counting objects: 100% (1971/1971), done.
remote: Compressing objects: 100% (423/423), done.
remote: Total 10050 (delta 1697), reused 1560 (delta 1535), pack-reused 8079 (from 1)
Receiving objects: 100% (10050/10050), 2.27 MiB | 2.82 MiB/s, done.
Resolving deltas: 100% (6297/6297), done.
traversaro@IITBMP014LW012:~$ cd jaxsim/
traversaro@IITBMP014LW012:~/jaxsim$ pixi run -e tasks-gpu --locked tests
 WARN The feature 'style' is defined but not used in any environment
 WARN The feature 'testing' is defined but not used in any environment
 WARN The feature 'viz' is defined but not used in any environment
 WARN The feature 'all' is defined but not used in any environment
  × lock-file not up-to-date with the project
~~~

and it fails with " × lock-file not up-to-date with the project". Probably the `pixi.toml` have been modified without updating the lock?

xela-95, 2024-11-20:
I think that is due to the update of `pyproject.toml` in 51ab0127b69410b0f42e7d5d3a8a18d265d7e8d4 in which we added the `trimesh` dependency after the last update of the lockfile action ran in #285

xela-95, 2024-11-20:
I ran the Pixi action to update the lockfile but we are getting a failing test: https://github.com/ami-iit/jaxsim/actions/runs/11929601986/job/33248711299

xela-95, 2024-11-20:
> I ran the Pixi action to update the lockfile but we are getting a failing test: https://github.com/ami-iit/jaxsim/actions/runs/11929601986/job/33248711299

Addressing this in #293

flferretti, 2024-12-04:
Since from `pixi` v0.39.0 the lockfile format has been changed, `pixi-diff-to-markdown` is not able to generate a diff anymore. We can either put an upper bound to the `pixi` version as in https://github.com/ami-iit/jaxsim/commit/28441a09fe93ae9d9f8a0467b54ec3b44780b528 or manually update the lockfile. Yet, I'm not sure whether the second option is backward compatible. WDYT @traversaro @xela-95?

traversaro, 2024-12-04:
> Yet, I'm not sure whether the second option is backward compatible. WDYT @traversaro @xela-95?

It is not. Once you update to pixi 0.39.0, the generated lock file will not be compatible with older pixi. Do you think this is a problem? See https://github.com/prefix-dev/pixi/issues/2278 for a related issue.

xela-95, 2024-12-04:
In theory they should have fixed this in https://github.com/pavelzw/pixi-diff-to-markdown/pull/41

flferretti, 2024-12-04:
> Do you think this is a problem? 

Definitely not, but users should be aware of this somehow. I started working on https://github.com/prefix-dev/pixi/issues/2278 in [flferretti/pixi@add_pixi_version](https://github.com/flferretti/pixi/tree/add-pixi-version), but I'm not quite sure if it's worth to spend time on this.

traversaro, 2024-12-04:
> Definitely not, but users should be aware of this somehow.

I think just updating the README to report the pixi minimum version I think it is sufficient.

PR #306 (2024-12-04): Update `pixi` lockfile
Closes #292 

# Explicit dependencies

|Dependency|Before|After|Change|Explicit|Environments|
|-|-|-|-|-|-|
|jaxlie|1.4.1|1.4.2|Patch Upgrade|true|{default, tasks-cpu} on {osx-64, osx-arm64}<br/>*all envs* on linux-64|
|mujoco|3.2.5|3.2.6|Patch Upgrade|true|{default, tasks-cpu} on *all platforms*|
|pytest|8.3.3|8.3.4|Patch Upgrade|true|tasks-cpu on *all platforms*<br/>tasks-gpu on linux-64|
|rod|0.3.3|0.3.4|Patch Upgrade|true|{default, tasks-cpu} on {osx-64, osx-arm64}<br/>*all envs* on linux-64|
|trimesh|4.5.2|4.5.3|Patch Upgrade|true|{default, tasks-cpu} on {osx-64, os [... cut]
closed 2024-12-04 by flferretti