gbionics/jaxsim #293: Test `test_simulation_with_relaxed_rigid_contacts` failing

xela-95, 2024-11-20:
The test in `tests/test_simulations.py::test_simulation_with_relaxed_rigid_contacts` is failing with the following error:

```python
 pytest -vv tests/test_simulations.py::test_simulation_with_relaxed_rigid_contacts                                                                                                                                 
======================================================================================================================================= test session starts =======================================================================================================================================
platform linux -- Python 3.13.0, pytest-8.3.3, pluggy-1.5.0 -- /home/acroci/mambaforge/envs/jaxsim/bin/python3.13
cachedir: .pytest_cache
rootdir: /home/acroci/repos/jaxsim
configfile: pyproject.toml
plugins: icdiff-0.9
collected 1 item                                                                                                                                                                                                                                                                                  

tests/test_simulations.py::test_simulation_with_relaxed_rigid_contacts FAILED                                                                                                                                                                                                               [100%]

============================================================================================================================================ FAILURES =============================================================================================================================================
___________________________________________________________________________________________________________________________ test_simulation_with_relaxed_rigid_contacts ___________________________________________________________________________________________________________________________

jaxsim_model_box = JaxSimModel(model_name='box', time_step=Array(0.001, dtype=float64))

    def test_simulation_with_relaxed_rigid_contacts(
        jaxsim_model_box: js.model.JaxSimModel,
    ):
    
        model = jaxsim_model_box
    
        with model.editable(validate=False) as model:
    
            model.contact_model = jaxsim.rbda.contacts.RelaxedRigidContacts.build(
                terrain=model.terrain,
                parameters=jaxsim.rbda.contacts.RelaxedRigidContactsParams.build(mu=0.001),
                solver_options={"tol": 1e-3},
            )
            # Enable a subset of the collidable points.
            enabled_collidable_points_mask = np.zeros(
                len(model.kin_dyn_parameters.contact_parameters.body), dtype=bool
            )
            enabled_collidable_points_mask[[0, 1, 2, 3]] = True
            model.kin_dyn_parameters.contact_parameters.enabled = tuple(
                enabled_collidable_points_mask.tolist()
            )
    
        assert np.sum(model.kin_dyn_parameters.contact_parameters.enabled) == 4
    
        # Initialize the maximum penetration of each collidable point at steady state.
        # This model is quasi-rigid, so we expect (almost) no penetration.
        max_penetration = 0.000
    
        # Check jaxsim_model_box@conftest.py.
        box_height = 0.1
    
        # Build the data of the model.
        data_t0 = js.data.JaxSimModelData.build(
            model=model,
            base_position=jnp.array([0.0, 0.0, box_height * 2]),
            velocity_representation=VelRepr.Inertial,
            # For this contact model, the following method is practically no-op.
            # Let's leave it there for consistency and to make sure that nothing
            # gets broken if it is updated in the future.
            contacts_params=js.contact.estimate_good_contact_parameters(
                model=model,
            ),
        )
        # ===========================================
        # Run the simulation and test the final state
        # ===========================================
    
        data_tf = run_simulation(model=model, data_t0=data_t0, dt=0.001, tf=1.0)
    
        # With this contact model, we need to slightly increase the tolerances.
>       assert data_tf.base_position()[0:2] == pytest.approx(
            data_t0.base_position()[0:2], abs=0.000_100
        )
E       assert equals failed
E         Array([-0.00019, -0.00004], dtype=float64)  approx([0.0 ± 1.0e-04, 0.0 ± 1.0e-04])

tests/test_simulations.py:402: AssertionError
```

@flferretti do you think it is indicating some regression in the relaxed rigid contact model behavior or we could just adapt the tolerance of the assert of the base position (or, alternatively, try to reduce the `dt` of the simulation or increase the simulation time of the test)?

flferretti, 2024-11-20:
I initially thought it was related to the fact that `optax` has been recently updated to [v0.2.4](https://github.com/google-deepmind/optax/releases/tag/v0.2.4), but apparently that version is not yet on conda-forge. I'm trying to reproduce locally

flferretti, 2024-11-20:
I opened https://github.com/conda-forge/optax-feedstock/pull/23 to see if udpating `optax` helps

traversaro, 2024-11-20:
> I opened [conda-forge/optax-feedstock#23](https://github.com/conda-forge/optax-feedstock/pull/23) to see if updating `optax` helps

To check if the test pass or now with the latest optax, you can also temporarly install the latest optax either from source or from PyPI.