gbionics/jaxsim #317: Document the operations nvolved in creating a JaxSimModel object

xela-95, 2024-12-12:
In this issue, I want to document what I find out and try to identify how to improve the readability of the code.

1. We start by calling the `build_from_model_description` class method by passing a path to an SDF (or URDF) file.
[jaxsim.api.model.JaxSimModel.build_from_model_description](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/api/model.py#L100-L114)

1.1. In `JaxSimModel.build_from_model_description()` [we create an intermediate_description](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/api/model.py#L148-L150) of type [`jaxsim.parsers.descriptions.ModelDescription`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/parsers/descriptions/model.py#L15-L16):
```python
        intermediate_description = jaxsim.parsers.rod.build_model_description(
            model_description=model_description, is_urdf=is_urdf
        )
```

1.1.1.[`jaxsim.parsers.rod.build_model_description()`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/parsers/rod/parser.py#L362-L365) creates an instance of ModelDescription by [first parsing the data in the SDF model](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/parsers/rod/parser.py#L380-L382)
```python
    sdf_data = extract_model_data(
        model_description=model_description, model_name=None, is_urdf=is_urdf
    )
```

1.1.1.1. [`extract_model_data()`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/parsers/rod/parser.py#L36-L40) returns an object of type `jaxsim.parsers.rod.SDFData` by first creating a `rod.Sdf` model using the `rod` library, then by parsing all the elements composing the model (links, joints, frames, etc.) and applying some assumptions on how they should be described. 

For example, it checks that all joint poses are expressed wrt the parent link,  that each link pose is expressed wrt their parent joint and processing fixed-base models in a custom way. Finally it creates an instance of `SDFData` and returns it.

1.1.2. Back to `jaxsim.parsers.rod.build_model_description()`, we then create a ModelDescription instance by calling :
```python
    graph = descriptions.ModelDescription.build_model_from(
        name=sdf_data.model_name,
        links=sdf_data.link_descriptions,
        joints=sdf_data.joint_descriptions,
        frames=sdf_data.frame_descriptions,
        collisions=sdf_data.collision_shapes,
        fixed_base=sdf_data.fixed_base,
        base_link_name=sdf_data.base_link_name,
        model_pose=sdf_data.model_pose,
        considered_joints=[
            j.name
            for j in sdf_data.joint_descriptions
            if j.jtype is not descriptions.JointType.Fixed
        ],
    )
```
1.1.2.1. [`ModelDescription.build_model_from`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/parsers/descriptions/model.py#L34-L45) first builds a `KinematicGraph` instance with:
```python
        # Create the full kinematic graph.
        kinematic_graph = KinematicGraph.build_from(
            links=links,
            joints=joints,
            frames=frames,
            root_link_name=base_link_name,
            root_pose=model_pose,
        )
```

1.1.2.1.1. [`KinematicGraph`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/parsers/kinematic_graph.py#L66) is a class that stores the kinematicGraph (called ConnectivityGraph in Featherstone book) in which the nodes of the graph are the links of the model and the edges are the joints.

[`KinematicGraph.build_from()`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/parsers/kinematic_graph.py#L161-L168) creates the graph by delegating a lot of the work to the [`create_graph`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/parsers/kinematic_graph.py#L221-L234) function. In this function a lot of of work is done, among which there is the selection of unconnected links, frames and joints that are excluded from the graph, and the connection of the nodes and arcs of the graph by assigning parents and children to links and joints.

Back to `KinematicGraph.build_from()`, it reduces the `kinematic_graph` by using only the `considered_joints` we passed:
```python
        # Reduce the graph if needed.
        if considered_joints is not None:
            kinematic_graph = kinematic_graph.reduce(
                considered_joints=considered_joints
            )
```
and then we create the `ModelDescription` object (that inherits from `KinematicGraph`) by passing to it the graph.

1.2. Back to `JaxSimModel.build_from_model_description()`, we have a now an `intermediate_description` object, then we reduce it by lumping links together for the joints we are not interested into:
```python
        # Note: this procedure assigns a zero position to all joints not considered.
        if considered_joints is not None:
            intermediate_description = intermediate_description.reduce(
                considered_joints=considered_joints
            )
``` 

1.3. Finally it calls the `build()` function of `JaxSimModel` by passing to it the new intermediate_description object.

1.3.1. The [`JaxSimModel.build()`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/api/model.py#L175-L187) is a factory method that creates an instance of JaxSimModel, which among other things, takes the `intermediate_description` and pass it to the build method of `jaxsim.api.kin_dyn_parameters.KynDynParameters`.
1.3.1.1. [`KynDynParameters.build()`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/api/kin_dyn_parameters.py#L61-L62) builds an instance of `KynDynParameters` by taking the `intermediate_description` (of type `ModelDescription`) we passed as argument, and creating a list of `LinkParameters`, `JointParameters`, `FrameParameters` (storing their index, among kinematic and dynamic parameters), creating the parent array `λ(i)` and the support parent array `κ(i)` that will be used in RBDAs.

xela-95, 2024-12-12:
1. We start by calling the `build_from_model_description` class method by passing a path to an SDF (or URDF) file.
[jaxsim.api.model.JaxSimModel.build_from_model_description](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/api/model.py#L100-L114)

1.1. In `JaxSimModel.build_from_model_description()` [we create an intermediate_description](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/api/model.py#L148-L150) of type [`jaxsim.parsers.descriptions.ModelDescription`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/parsers/descriptions/model.py#L15-L16):
```python
        intermediate_description = jaxsim.parsers.rod.build_model_description(
            model_description=model_description, is_urdf=is_urdf
        )
```

1.1.1.[`jaxsim.parsers.rod.build_model_description()`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/parsers/rod/parser.py#L362-L365) creates an instance of ModelDescription by [first parsing the data in the SDF model](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/parsers/rod/parser.py#L380-L382)
```python
    sdf_data = extract_model_data(
        model_description=model_description, model_name=None, is_urdf=is_urdf
    )
```

1.1.1.1. [`extract_model_data()`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/parsers/rod/parser.py#L36-L40) returns an object of type `jaxsim.parsers.rod.SDFData` by first creating a `rod.Sdf` model using the `rod` library, then by parsing all the elements composing the model (links, joints, frames, etc.) and applying some assumptions on how they should be described. 

For example, it checks that all joint poses are expressed wrt the parent link,  that each link pose is expressed wrt their parent joint and processing fixed-base models in a custom way. Finally it creates an instance of `SDFData` and returns it.

1.1.2. Back to `jaxsim.parsers.rod.build_model_description()`, we then create a ModelDescription instance by calling :
```python
    graph = descriptions.ModelDescription.build_model_from(
        name=sdf_data.model_name,
        links=sdf_data.link_descriptions,
        joints=sdf_data.joint_descriptions,
        frames=sdf_data.frame_descriptions,
        collisions=sdf_data.collision_shapes,
        fixed_base=sdf_data.fixed_base,
        base_link_name=sdf_data.base_link_name,
        model_pose=sdf_data.model_pose,
        considered_joints=[
            j.name
            for j in sdf_data.joint_descriptions
            if j.jtype is not descriptions.JointType.Fixed
        ],
    )
```
1.1.2.1. [`ModelDescription.build_model_from`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/parsers/descriptions/model.py#L34-L45) first builds a `KinematicGraph` instance with:
```python
        # Create the full kinematic graph.
        kinematic_graph = KinematicGraph.build_from(
            links=links,
            joints=joints,
            frames=frames,
            root_link_name=base_link_name,
            root_pose=model_pose,
        )
```

1.1.2.1.1. [`KinematicGraph`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/parsers/kinematic_graph.py#L66) is a class that stores the kinematicGraph (called ConnectivityGraph in Featherstone book) in which the nodes of the graph are the links of the model and the edges are the joints.

[`KinematicGraph.build_from()`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/parsers/kinematic_graph.py#L161-L168) creates the graph by delegating a lot of the work to the [`create_graph`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/parsers/kinematic_graph.py#L221-L234) function. In this function a lot of of work is done, among which there is the selection of unconnected links, frames and joints that are excluded from the graph, and the connection of the nodes and arcs of the graph by assigning parents and children to links and joints.

Back to `KinematicGraph.build_from()`, it reduces the `kinematic_graph` by using only the `considered_joints` we passed:
```python
        # Reduce the graph if needed.
        if considered_joints is not None:
            kinematic_graph = kinematic_graph.reduce(
                considered_joints=considered_joints
            )
```
and then we create the `ModelDescription` object (that inherits from `KinematicGraph`) by passing to it the graph.

1.2. Back to `JaxSimModel.build_from_model_description()`, we have a now an `intermediate_description` object, then we reduce it by lumping links together for the joints we are not interested into:
```python
        # Note: this procedure assigns a zero position to all joints not considered.
        if considered_joints is not None:
            intermediate_description = intermediate_description.reduce(
                considered_joints=considered_joints
            )
``` 

1.3. Finally it calls the `build()` function of `JaxSimModel` by passing to it the new intermediate_description object.

1.3.1. The [`JaxSimModel.build()`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/api/model.py#L175-L187) is a factory method that creates an instance of JaxSimModel, which among other th [... cut ...]
1.3.1.1. [`KynDynParameters.build()`](https://github.com/ami-iit/jaxsim/blob/c896b48a30da465486bcfe2069d6f236fba4460d/src/jaxsim/api/kin_dyn_parameters.py#L61-L62) builds an instance of `KynDynParameters` by taking the `intermediate_description` (of type `ModelDescription`) we passed as argument, and creating a list of `LinkParameters`, `JointParameters`, `FrameParameters` (storing their index, among kinematic and dynamic parameters), creating the parent array `λ(i)` and the support parent array `κ(i)` that will be used in RBDAs.

S-Dafarra, 2024-12-16:
Is the end user supposed to follow all these steps? It seemed utterly convoluted in my opinion

xela-95, 2024-12-16:
> Is the end user supposed to follow all these steps? It seemed utterly convoluted in my opinion

No in the end this is too complex and with too much detailed for the user. I wanted to use this PR as future reference for developers working on this features of Jaxsim and as point of discussion for future improvements.

xela-95, 2024-12-17:
We have documented the operations involoved in the creation of a JaxSim model, closing for now.

closed 2024-12-17 by xela-95