gbionics/jaxsim #386: Investigate wrong behaviour in single-stance phases

lorycontixd, 2025-03-04:
The solution of the equations of motion generate the trajectory of the hip point mass as shown in the following figure:
> The figure shows the x,y,z coordinates of the point mass at the hip on the left, and the state variables on the right. It performs only one step.

[image]

First of all, the y-axis represents the sideways movement of the walker along the inclined plane, while the x-axis represents the forward direction of movement (inertial to the slope).

The initial state conditions for the above simulation are:
$\theta_1 = 0.1,\quad \omega_1 = 0.0$
$\omega_2 = 0.1,\quad \omega_2 = 0.0$
$\omega = 0.1\quad \omega = 0.0;$

The dynamics shown above at least has the 2 following problems:
- **Inverted motion:** The value that keeps increasing (or decreasing) should be the forward axis (x), while the y axis should oscillate around the equilibrium $y=0$. Ignoring the divergence at a certain point, in this simulation the x values oscillates (and not even around $x=0$), while the y value keeps decreasing.
- **Divergence:** The expected behaviour of a step would be that the oscillatory movement completes an oscillation (returns to the initial point after one step), but in this case there is an unexpected change in direction which makes the step diverge.
-

flferretti, 2025-03-04:
Hi @lorycontixd, maybe you wanted to open this in a different repo? I'm missing the relationship with JaxSim

lorycontixd, 2025-03-04:
That's where it went.. I couldn't find it in my element 🤦‍♂️
Thanks

closed 2025-03-04 by lorycontixd