gbionics/jaxsim #393: Joint order does not follow the joint name list when reducing the model

CarlottaSartore, 2025-03-27:
In Jaxsim, eventhough we pass a joint name list when we reduce the model, the joint order does not follow the passed list but the order in which the kinematic chain is created in the associated urdf/sdf file. To inspect the correct joint name list one should use the `model` method `joint_names()`.

It happens few times that this thing  create confusion, we should find a way either to correctly allert the user about this or  use the joint order passed by the user 

C.C. @ami-iit/darwin

traversaro, 2025-03-27:
Is this behavior clear in the model reduction documentation? Just to provide a bit of context for external users, the reason why most users in AMI expect that model reducer API both specify which joints are part of the reduced model and their order is that iDynTree API provides that, while most other dynamics libraries use what Featherstone call "normal joint ordering", where the order of the joints depends on the chosen base.

CarlottaSartore, 2025-03-27:
> Is this behavior clear in the model reduction documentation? Just to provide a bit of context for external users, the reason why most users in AMI expect that model reducer API both specify which [...]

I think  in the documentation gives for granted that the order will not changes, this with the  AMI bias had lead (and most likely will) be source of errors, as a first step I think we can improve the documentation, but I am tempted to add also a warning. The problem I see is that currently there are a lot of prints when the model is loaded, thus an additional print most likely will get lost  
https://github.com/ami-iit/jaxsim/blob/2c62beba2de080bedb6d673f1483f820ac7da430/src/jaxsim/parsers/descriptions/model.py#L156-L188

flferretti, 2025-03-27:
I agree with adding a logging in the `js.model.reduce`

> there are a lot of prints when the model is loaded

This is due to the fact that the logging level is set to DEBUG when JaxSim is installed in editable mode

CarlottaSartore, 2025-03-27:
> This is due to the fact that the logging level is set to DEBUG when JaxSim is installed in editable mode

Since internally in A,MI most user use  Jaxsim in editable mode, we could decrease the things printed, so that more critical things, as the joint order, do not get lost.

flferretti, 2025-03-27:
> > This is due to the fact that the logging level is set to DEBUG when JaxSim is installed in editable mode
> 
> Since internally in A,MI most user use Jaxsim in editable mode, we could decrease the things printed, so that more critical things, as the joint order, do not get lost.

I agree with that. We could move some logs from DEBUG to TRACE level, especially when the links are lumped

flferretti, 2025-04-03:
> > This is due to the fact that the logging level is set to DEBUG when JaxSim is installed in editable mode
> 
> Since internally in A,MI most user use Jaxsim in editable mode, we could decrease the things printed, so that more critical things, as the joint order, do not get lost.

Done in https://github.com/ami-iit/jaxsim/tree/fix/logging_level

flferretti, 2025-04-17:
I agree, it should be closed by #398.

Feel free to close this, thank you!

closed 2025-04-17 by CarlottaSartore