gbionics/jaxsim #410: Default Joint positions may be out of limits

younik, 2025-04-14:
We currently initialize joint positions as zeros by default:

https://github.com/ami-iit/jaxsim/blob/9d55da69f3ed8a1db28dfe2fb63371b7482d79d8/src/jaxsim/api/data.py#L125-L133

However, some joints have limits that don't include zero (for example, stickbot shoulder).
We may want to initialize them differently, for example in the mean of limits.

CarlottaSartore, 2025-04-14:
Nice catch @younik  ! I think we could do that but it had sense only if the joint limits are activated !  There should be a flag associated to it if I am not wrong, we could check if the joint limits are activated we  initialize to zero or the limits closer to zedro, while if they are not  activated we initialize to zero. 

I would prefer to keep anyway a relation with the zeros joint position since it is usually the neutral position

younik, 2025-04-14:
> I would prefer to keep anyway a relation with the zeros joint position since it is usually the neutral position

Right, it can also be just a clamp after zero init.