gbionics/jaxsim #427: Issue in updating hw metadata in model in which not all links are parametrizable

xela-95, 2025-05-16:
## Description
When trying to update the hw parameters of a model in which some of the links have been skipped (since they are not supported, for example due to multiple visuals), we get an error. We should account for the fact that the parametrizable links are always a subset of the links of a model (one of the edge cases is that the two sets coincide but this is only one possibility) when handling them.

 
## Repro

```python
import os
from pathlib import Path

os.environ["JAXSIM_LOGGING_LEVEL"] = "debug"

import jax.numpy as jnp
import jaxsim.api as js

urdf_file = <path-to-urdf>

model = js.model.JaxSimModel.build_from_model_description(
    model_description=urdf_file,
    time_step=0.001,
    is_urdf=True,
)

# model = js.model.reduce(model=model, considered_joints=JOINT_NAMES)

base_position = jnp.array([0.0, 0.0, 1.5], dtype=jnp.float32)

data = js.data.JaxSimModelData.build(
    model=model,
    base_position=base_position,
)

print(f"Link names: \n{model.link_names()}")

print(f"Joint names: \n{model.joint_names()}")

# Define the link used in the loss function.
link_name = "torso_1"
link_idx = js.link.name_to_idx(model, link_name=link_name)

# Define the initial hardware parameters (scaling factors).
initial_dims = jnp.ones(
    (
        model.number_of_links(),
        3,
    )
)  # Initial dimensions (1.0 for all links).
initial_density = jnp.ones(
    (model.number_of_links(),)
)  # Initial density (1.0 for all links).

s = jnp.ones(
    (
        model.number_of_links(),
        3,
    )
)  # Initial dimensions (1.0 for all links).

s = s.at[link_idx].set(jnp.array([2.0, 1.0, 1.0]))  # manual scaling

scaling_factors = js.kin_dyn_parameters.ScalingFactors(
    dims=s,
    density=initial_density,
)

# Update the model with the optimized scaling factors.
optimized_model: js.model.JaxSimModel = js.model.update_hw_parameters(
    model=model, scaling_factors=scaling_factors
)

[... 924 lines cut ...]
     33     with jax.named_scope(name or fn.__name__):
---> 34         return fn(*args, **kwargs)

File ~/repos/jaxsim/src/jaxsim/api/model.py:2337, in update_hw_parameters(model, scaling_factors)
   2334 hw_link_metadata: HwLinkMetadata = kin_dyn_params.hw_link_metadata
   2336 # Apply scaling to hw_link_metadata using vmap
-> 2337 updated_hw_link_metadata = jax.vmap(HwLinkMetadata.apply_scaling)(
   2338     hw_link_metadata, scaling_factors
   2339 )
   2341 # Compute mass and inertia once and unpack the results
   2342 m_updated, I_com_updated = jax.vmap(HwLinkMetadata.compute_mass_and_inertia)(
   2343     updated_hw_link_metadata
   2344 )

    [... skipping hidden 2 frame]

File ~/repos/component_darwin/.pixi/envs/default/lib/python3.13/site-packages/jax/_src/api.py:1091, in _mapped_axis_size(fn, tree, vals, dims, name)
   1089   else:
   1090     msg.append(f"  * some axes ({ct} of them) had size {sz}, e.g. axis {ax} of {ex};\n")
-> 1091 raise ValueError(''.join(msg)[:-2])

ValueError: vmap got inconsistent sizes for array axes to be mapped:
  * most axes (7 of them) had size 16, e.g. axis 0 of argument hw_metadata[<flat index 0>] of type int64[16];
  * some axes (2 of them) had size 23, e.g. axis 0 of argument scaling_factors[<flat index 0>] of type float64[23,3]
```

flferretti, 2025-05-20:
@xela-95 maybe #432 fixes this

CarlottaSartore, 2025-09-17:
What is the status of this issue?